MarlinFirmware/Configurations #427: [BUG] Anycubic kossel linear plus #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."

pasikov123, 2021-02-25:
When trying to compile and load the marlin 2.0.7.2 build for Anycubic kossel linear plus with a btt skr 1.4 board using visual studio code, we get the error #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."

My configurations
sanicubic kassel linear plus
marlin 2.0.7.2
MOTHERBOARD BOARD_BTT_SCR_V1_4
#define ANYCUBIC_PROBE_VERSION 2

Steps to play back
1-Downloaded marlin version 2.0.7.2.
2-Settings in Configuration. h
3-Build

Expected behavior: successful build/upload

Actual behavior: # error "Z_MIN_PROBE_USERS_Z_MIN_ENDSTOP_PIN requires Z_MIN_PIN definition."

Additional information
The problem is solved by updating the pins_BT_SCR_V1_4.h file from

[pins_BTT_SKR_V1_4.zip](https://github.com/MarlinFirmware/Configurations/files/6044373/pins_BTT_SKR_V1_4.zip)

ellensp, 2021-02-27:
skr 1.4 only has 3 endstops
X_STOP
Y_STOP
Z_STOP

you have set them all to 
#define X_HOME_DIR 1 
#define Y_HOME_DIR 1
#define Z_HOME_DIR 1

so all the endstops are setup as MAX endstops so you can home to max.

There is no Z_MIN_PIN

you have changed the pin definitions
redefining E0DET E1DET and PWRDET as endstop. which they is not their default fuction
This will break all users that are using the pins for their intended function.

ellensp, 2021-02-27:
On this board you should not enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN but instead use Z_MIN_PROBE_PIN to the pin you are using.

closed 2021-02-27 by ellensp

pasikov123, 2021-02-27:
Z_MIN_PROBE_ASUS_Z_MIN_ENDSTOP_PIN is enabled by default for the kossel configuration

 #if ANYCUBIC_PROBE_VERSION > 0
   #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 #endif

ellensp, 2021-02-27:
for the stock controller, you changed the controller, invalidating all assumption in the config

pasikov123, 2021-02-27:
SKR 1.4 is not suitable for Delta because there is no pin for PROBE_VERSION

ellensp, 2021-02-27:
No not at all.. Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is not suitable for a skr1.4 on a delta

ellensp, 2021-02-27:
[image]

pasikov123, 2021-02-27:
Then it turns out need to uncomment #define Z_MIN_PROBE_PIN P1_00 ?

ellensp, 2021-02-27:
If you wish to force it to use the PWRDET plug, yes.

pasikov123, 2021-03-01:
By specifying P1_00, the error did not disappear. Tell me what else you need to enable to make ANYCUBIC_PROBE work?