MarlinFirmware/Configurations #428: Building errors in Auto Build Marlin Firmware [VSCode] due to configuration header for ender 3 v1 files

ZachR65, 2021-02-26:
n file included from Marlin\src\module\../inc/MarlinConfig.h:47:0,
                 from Marlin\src\module\motion.h:31,
                 from Marlin\src\module\motion.cpp:27:
Marlin\src\module\motion.cpp:151:16: error: 'HOMING_FEEDRATE_XY' was not declared in this scope
     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),
                ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^~~~
Marlin\src\module\motion.cpp:151:16: note: suggested alternative: 'HOMING_FEEDRATE_MM_M'
     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),
Compiling .pio\build\sanguino1284p\src\src\sd\SdFile.cpp.o
                ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
Compiling .pio\build\sanguino1284p\src\src\sd\SdVolume.cpp.o
                                           ^~~~
Marlin\src\module\motion.cpp:151:48: error: 'HOMING_FEEDRATE_XY' was not declared in this scope
     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),
                                                ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^~~~
Marlin\src\module\motion.cpp:151:48: note: suggested alternative: 'HOMING_FEEDRATE_MM_M'
     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),
                                                ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^~~~
Marlin\src\module\motion.cpp:153:14: error: 'HOMING_FEEDRATE_Z' was not declared in this scope
   MMM_TO_MMS(HOMING_FEEDRATE_Z)
              ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^~~~
Marlin\src\module\motion.cpp:153:14: note: suggested alternative: 'HOMING_FEEDRATE_MM_M'
   MMM_TO_MMS(HOMING_FEEDRATE_Z)
              ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^~~~
Marlin\src\module\motion.cpp: In function 'void do_z_clearance(const float&, bool, bool, bool)':
Marlin\src\module\motion.cpp:513:100: error: 'HOMING_FEEDRATE_Z' was not declared in this scope
   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));
                                                                                                    ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^~~~
Marlin\src\module\../inc/../core/macros.h:196:29: note: in expansion of macro 'THIRD'
 #define ___TERN(P,V...)     THIRD(P,V)              // If first argument has a comma, A. Else B.
                             ^~~~~
Marlin\src\module\../inc/../core/macros.h:195:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:194:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION converted to '0' or '1'
                             ^~~~~
Marlin\src\module\motion.cpp:513:60: note: in expansion of macro 'TERN'
   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));
                                                            ^~~~
Marlin\src\module\motion.cpp:513:100: note: suggested alternative: 'HOMING_FEEDRATE_MM_M'
   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));
                                                                                                    ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^~~~
Marlin\src\module\../inc/../core/macros.h:196:29: note: in expansion of macro 'THIRD'
 #define ___TERN(P,V...)     THIRD(P,V)              // If first argument has a comma, A. Else B.
                             ^~~~~
Marlin\src\module\../inc/../core/macros.h:195:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:194:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION converted to '0' or '1'
                             ^~~~~
Marlin\src\module\motion.cpp:513:60: note: in expansion of macro 'TERN'
   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));
                                                            ^~~~
*** [.pio\build\sanguino1284p\src\src\module\motion.cpp.o] Error 1
========================================================================================= [FAILED] Took 12.16 seconds =========================================================================================
Environment    Status    Duration
-------------  --------  ------------
sanguino1284p  FAILED    00:00:12.163

rhapsodyv, 2021-02-26:
Attach your configs please.

ZachR65, 2021-02-26:
[CrealityV1.zip](https://github.com/MarlinFirmware/Configurations/files/6051573/CrealityV1.zip)

ZachR65, 2021-02-26:
I have watched so many videos, and read so many forums/ how to's but this is really frustrating. It doesnt happen to anyone in these videos. What am i doing wrong?

ellensp, 2021-02-27:
your using bugfix/import config files on marlin 2.0.7.2

ellensp, 2021-02-27:
for marlin 2.0.7.2 or 2.0.x use release 2.0.7.2 config files https://github.com/MarlinFirmware/Configurations/archive/release-2.0.7.2.zip

closed 2021-02-27 by ellensp