MarlinFirmware/Configurations #431: [BUG] Missing tag for configs for 2.0.7.2

marciot, 2021-03-02:
One of the unfortunate consequences of having the configuration files outside of the main Marlin repo is that it is necessary to find the appropriate configs for earlier releases. This is made difficult when the appropriate tags do not exist, such as for release 2.0.7.2

ManuelMcLure, 2021-03-02:
There is a `release-2.0.7.2` branch in the configuration repository - doesn't this cover the requirement for getting the right configuration for 2.0.7.2?

https://github.com/MarlinFirmware/Configurations/tree/release-2.0.7.2

marciot, 2021-03-02:
I suppose it would. I didn’t know about that. Thank you.

ellensp, 2021-03-02:
In the Marlin you downloaded is a config directory. In this directory is a readme file which lists where to get the correct example configurations for your version of the code.

closed 2021-03-02 by ellensp