MarlinFirmware/Configurations #446: [FR] (configuration request title)

xtronics, 2021-03-26:
### Description

CR10S_Pro  (not to be confused with the CR10S or CR10S_Pro_V2 )

* Manufacturer: Creality
* Model: CR10S_Pro
* Motherboard:  I think this is a 2560

#### Additional Information
Creality has source code - but it won't build via makefile - nor will it build via Arduino IDE - the Configuration.h file is not compatible with the newer Marlin -  There are way to many things in the new configuration.h file that I don't know what to set to.

I think it is possible that the LCD that comes with the CR10S_Pro is not compatible with the current Marlin? (I don't know)  - even so - if there isn't a config - a folder with instead a readme telling owners they have a problem would be a good thing. 

As an aging compute guy - I found getting to the point of building this on Linux extremely tedious - a simple bash scrip that gives a list of packages needed to be installed - and then runs a command to compile would be sweet.  

* Provide links to the requested printer/configuration.

ellensp, 2021-03-26:
Link to firmware source?, as it will provide clues.

ellensp, 2021-03-26:
Link to firmware https://forums.creality3dofficial.com/download/cr-series/cr-10s-pro/

xtronics, 2021-03-26:
That link is older - there is a version at https://file2-cdn.creality.com//website/bc8a9040-e119-47f3-90fb-f87f8c0afa3c

This link may be temporary - if it dosen't work - try https://www.creality.com/download - go to the second page of CR printers - you will see CR10S_pro..

ellensp, 2021-03-26:
please update the tittle.

ellensp, 2021-03-26:
Some motherboard images
[image]
[image]

ellensp, 2021-03-26:
In the base
[image]

ellensp, 2021-03-26:
TFT seems to be same as one as in in CR-10S / Ender5+
[image]

xtronics, 2021-03-27:
It has auto leveling features (5 zone manual adjust and a 36? point auto level routine) 
and filament detector if that helps? 
https://preview.redd.it/g54yf1xuo8g21.png?width=2560&format=png&auto=webp&s=79cd8559ca78639a782abc415157683318708693

thinkyhead, 2022-04-10:
Duplicate of #649

closed 2022-04-10 by thinkyhead