MarlinFirmware/Configurations #448: Driver setting

MarioPerini, 2021-03-28:
https://github.com/MarlinFirmware/Configurations/blob/dd4de330913ae7e416a66a33e4b76a5fe8ece807/config/examples/Velleman/K8400/Single-head/Configuration.h#L739

sure here? on my board i find drv8825 drivers.

thinkyhead, 2021-04-25:
We rely on electronics owners to tell us what is going on. If there is a Facebook or other user group, please compare notes with others and report back what kind of drivers these boards have, both older and newer ones, and then the configurations may be corrected.

thisiskeithb, 2022-08-05:
Based on Velleman's [K8400 3D Printer Technical Doc](https://github.com/MarlinFirmware/Configurations/files/9271489/K8400.3D.Printer.Technical.Doc.pdf), I submitted https://github.com/MarlinFirmware/Configurations/pull/793 to address this.

closed 2022-08-05 by thisiskeithb