MarlinFirmware/Configurations #451: ender 3 MAX configuration bug

spooky79, 2021-03-29:
ender 3 MAX has a Creality 4.2.2 mainboard with X,Y TMC2208 and  Z,E A4988 , there is an error in the configuration file

Vertabreak, 2021-03-29:
suggest submitting a pull request with the noted changed for review.

thisiskeithb, 2021-03-29:
> suggest submitting a pull request with the noted change for review.

There’s likely multiple versions of this printer that includes different boards just like the rest of Creality’s lineup.

Vertabreak, 2021-03-29:
from what i understood as long as a company produced the printer, marlin would accept a variant config for it.

thisiskeithb, 2021-03-29:
> from what i understood as long as a company produced the printer, marlin would accept a variant config for it.

Of course.

I’m stating that instead of declaring a config as incorrect and changing it, create another variant since Creality is known to do that.

Vertabreak, 2021-03-29:
yea and there not the only one making variants all over its very troublesome.

spooky79, 2021-03-29:
ender 3 max has only one variant mainboard, these are mainboard made especially for this printer, only the pre-sale test version had a standard mainboard with 4x a4988.  This was also confirmed Creality  in the message from them. At the moment there are no other options like  mainboard with 2x TMC2208 and 2x A4988.

moccalotto, 2021-04-30:
Also - **important** -   `#define BLTOUCH_SET_5V_MODE` is __enabled__ in the (advanced) config file.

I am far from sure that this is ok since it is disabled on all other config files for 422 motherboards (for instance ender 3v2). I have disabled this flag on my in own config as I'm not sure it is safe to try 5v

stacherman, 2022-06-17:
@spooky79 is correct, there is only one board config in terms of stepper motors. I am submitting a pull request to correct the configuration file.

thisiskeithb, 2022-07-25:
Fixed in https://github.com/MarlinFirmware/Configurations/pull/789

closed 2022-07-25 by thisiskeithb