MarlinFirmware/Configurations #453: [FR] (configuration request title)

Laurelinwen, 2021-04-03:
### Description

### Configuration Details

* Manufacturer: Geeetech
* Model: A10T
* Motherboard:Atmega GT2560 V4

#### Additional Information

* Provide links to the requested printer/configuration. 
* https://www.geeetech.com/geeetech-a10t-3-in-1-out-mixcolor-auto-leveling-3d-printer-p-1107.html
* https://www.geeetech.com/gt2560-v40-control-board-for-a10t-printer-p-1164.html
* https://www.geeetech.com/stepper-driver-a4988-with-heatsink-and-sticker-p-1007.html
* https://www.geeetech.com/titan-extruder-kit-for-a10m-a10t-a20m-a20t-a30m-a30t-p-1131.html
* https://www.geeetech.com/geeetech-a10-a10m-a10t-heatbed-kit-p-1104.html
* https://www.geeetech.com/download.html?download_id=49, the Firmware is .hex only.

Vertabreak, 2021-04-03:
this already exists.
https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Geeetech

Laurelinwen, 2021-04-03:
I will try it. Thank you so far.

closed 2021-04-04 by thisiskeithb