MarlinFirmware/Configurations #454: [FR] (configuration request title)

Laurelinwen, 2021-04-03:
### Description

### Configuration Details

* Manufacturer: Anycubic
* Model: Mega Zero 2.0
* Motherboard: Mega Zero 0.0.3 (Melzi-Board mit) ATMega 1284P U-TW

#### Additional Information

* Provide links to the requested printer/configuration.
* https://www.anycubic.com/collections/anycubic-mega-3d-printers/products/mega-zero-2-0-fdm-3d-printer
* https://github.com/ANYCUBIC-3D/MEGA_ZERO
* https://drucktipps3d.de/anycubic-mega-zero-2-0/

ellensp, 2021-04-07:
Please carefully test this PR https://github.com/MarlinFirmware/Configurations/pull/458
I do not have this machine to try it on.

ellensp, 2021-04-13:
Can you please try these Configs

closed 2021-04-14 by thinkyhead