MarlinFirmware/Configurations #455: [BUG] invalid conversion from 'const char*' to 'char* when compiling

valemaio2, 2021-04-05:
### Bug Description

_invalid conversion from 'const char*' to 'char*_ Error when compiling the latest bugfix for the Creality 4.2.2 board.

When compiling the latest Bugfix for the Ender 3 Pro, board 4.2.2, this error in VisualStudio occours:

```
Marlin\src\sd\cardreader.cpp: In static member function 'static void CardReader::write_command(const char*)':
Marlin\src\sd\cardreader.cpp:726:17: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
   char *begin = buf;
                 ^~~
   char *end = buf + strlen(buf) - 1;
               ~~~~~Compiling .pio\build\STM32F103RET6_creality\lib80c\U8glib-HAL\clib\u8g_dev_ssd1306_128x64.c.o
~~~~~~~~~~~~~^~~
Compiling .pio\build\STM32F103RET6_creality\lib80c\U8glib-HAL\clib\u8g_dev_ssd1306_64x48.c.o
*** [.pio\build\STM32F103RET6_creality\src\src\sd\cardreader.cpp.o] Error 1
```

### Configuration Files

[ZIP file with configuration.h, configuration_adv.h and cardreader.cpp](https://github.com/MarlinFirmware/Configurations/files/6257855/bug.zip)

### Steps to Reproduce

1. Download latest bugfix (commit 98d6c75) and latest bugfix example files (commit d93cffb)
2. Compile in VisualStudio
3. Get the error

**Expected behavior:**

Compiling a .bin firmware file

**Actual behavior:**

Error in the description above, no .bin file

#### Additional Information

None

ellensp, 2021-04-05:
duplicate of https://github.com/MarlinFirmware/Marlin/issues/21542#issuecomment-813333394

closed 2021-04-05 by ellensp

valemaio2, 2021-04-05:
> 
> 
> duplicate of [MarlinFirmware/Marlin#21542](https://github.com/MarlinFirmware/Marlin/issues/21542)

I confirm, after reverting the changes in 98d6c75, I managed to compile without errors.