MarlinFirmware/Configurations #460: [BUG] (short description)

fumanschuh, 2021-04-07:
Hi
I get the following error message when I try to run the firmware for the sapphire plus with bltouch. I use the configuration.h and configuration.adv.h for sapphire plus and only define lvgl ui and bltouch.

https://linksharing.samsungcloud.com/mardgvMCt4dz

ellensp, 2021-04-07:
board is a BOARD_MKS_ROBIN_NANO
your error says your incorrectly compiling for AVR
Set platfomio.ini default_envs = mks_robin_nano35 (or mks_robin_nano35_stm32 if bugfix)

closed 2021-04-07 by ellensp

ellensp, 2021-04-07:
For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

fumanschuh, 2021-04-07:
when I compile now it always comes HOMING_FEEDRATE_XY was not declared in this scope
Also for FEEDRATE Z and Z_PROBE_SPEED_FAST and slow

ellensp, 2021-04-07:
Because your using bugfix or import version of the config files on release version marlin. Those names and otehrs have changed in bugfix.