MarlinFirmware/Configurations #461: [en] ender 5 plus  working configuration files!

Helvensteijn29, 2021-04-07:
hi the configuration file is corrupt and doesn't compile (5 ERRORS)
ENDER 5 PLUS MARLIN 2.0

Already thanks

regards,

Michel

thisiskeithb, 2021-04-07:
Duplicate of https://github.com/MarlinFirmware/Configurations/issues/459

closed 2021-04-07 by thisiskeithb

Helvensteijn29, 2021-04-08:
thanks for the fast response, they are not active,
this is the compile response that I get 
already thanks in advance :-)

Marlin/src/lcd/extui/lib/dgus/origin/DGUSDisplayDef.cpp:310:1: error: no matches converting function 'GetVariable' to type 'void (*)(struct DGUS_VP_Variable&, void*)'
 };
 ^
In file included from Marlin/src/lcd/extui/lib/dgus/origin/DGUSDisplayDef.cpp:32:0:
Marlin/src/lcd/extui/lib/dgus/origin/../DGUSDisplay.h:76:15: note: candidate is: template<class T, void (* Setter)(float, T), T selector> static void DGUSDisplay::GetVariable(DGUS_VP_Variable&, void*)
   static void GetVariable(DGUS_VP_Variable &var, void *val_ptr) {
               ^~~~~~~~~~~
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_FileSelected(DGUS_VP_Variable&, void*)':
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp:47:22: error: 'filelist' was not declared in this scope
     if (touched_nr > filelist.count()) return;
                      ^~~~~~~~
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp:47:22: note: suggested alternative: 'filepos_t'
     if (touched_nr > filelist.count()) return;
                      ^~~~~~~~
                      filepos_t
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp:48:10: error: 'filelist' was not declared in this scope
     if (!filelist.seek(touched_nr)) return;
          ^~~~~~~~
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp:48:10: note: suggested alternative: 'filepos_t'
     if (!filelist.seek(touched_nr)) return;
          ^~~~~~~~
          filepos_t
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp:50:9: error: 'filelist' was not declared in this scope
     if (filelist.isDir()) {
         ^~~~~~~~
*** [.pio/build/mega2560/src/src/lcd/extui/lib/dgus/origin/DGUSDisplayDef.cpp.o] Error 1
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp:50:9: note: suggested alternative: 'filepos_t'
     if (filelist.isDir()) {
         ^~~~~~~~
         filepos_t
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp:65:87: error: 'filelist' was not declared in this scope
     HandleUserConfirmationPopUp(VP_SD_FileSelectConfirm, nullptr, PSTR("Print file"), filelist.filename(), PSTR("from SD Card?"), true, true, false, true);
                                                                                       ^~~~~~~~
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp:65:87: note: suggested alternative: 'filepos_t'
     HandleUserConfirmationPopUp(VP_SD_FileSelectConfirm, nullptr, PSTR("Print file"), filelist.filename(), PSTR("from SD Card?"), true, true, false, true);
                                                                                       ^~~~~~~~
                                                                                       filepos_t
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_StartPrint(DGUS_VP_Variable&, void*)':
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp:69:10: error: 'filelist' was not declared in this scope
     if (!filelist.seek(file_to_print)) return;
          ^~~~~~~~
Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp:69:10: note: suggested alternative: 'filepos_t'
     if (!filelist.seek(file_to_print)) return;
          ^~~~~~~~
          filepos_t
compilation terminated due to -fmax-errors=5.
*** [.pio/build/mega2560/src/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp.o] Error 1
========================================= [FAILED] Took 3.09 seconds =========================================

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:03.090
=================================== 1 failed, 0 succeeded in 00:00:03.090 ===================================
michelhelvensteijn@MacBook-Pro-van-Michel Marlin-bugfix-2.0-2.x %