MarlinFirmware/Configurations #470: [BUG] LCD12864只显示白屏  LCD12864 only displays white screen

dgpower, 2021-04-16:
12864白屏问题及解决方案
------------------------------------------------
我尝试用最新的marlin固件版本，包括最新bugifx和marlin2.0.7.2，发现只要我液晶显示器配置了中文菜单（zh_cn），刷新固件后液晶显示器上就是一篇空白，没有任何字符。
[image]

经过无数次的测试，我发现，只要不打开EEPROM功能，就可以正常显示

ellensp, 2021-04-16:
For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

dgpower, 2021-04-16:
[Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/6325083/Marlin.zip)

thinkyhead, 2021-04-16:
> the operating memory of the LCD12864 is full

That is unlikely. It stores nothing. Marlin controls it 100%.

But as you discovered, you can turn off features to get some SRAM back. It doesn't have to be EEPROM.

Anyway, version 2.0.8 will be released soon. It has many improvements.

closed 2021-04-16 by thinkyhead