MarlinFirmware/Configurations #473: Where to disable the silly confirmation screens

nay27uk, 2021-04-17:
Please what file / files do I open and edit and on what lines in them to disable the silly confirmation screens in marlin 2.0 I cant find it anywhere.

I want to disable the comfirmation pages that pop up on the LCD of the ender 3 when STOPING, PAUSING a print etc.

I dont want any confirmation pages at all just like the stock creality firmware.

I have enough of Windows as it is asking me to confirm crap all the time I dont need it on a 3D printer too anoying confirmations.

thisiskeithb, 2021-04-24:
> Where to disable the silly confirmation screens

Lots of places. Confirmation screens are not a globally configurable option.

You'll need to touch a lot of core files within Marlin, so run a project-wide search for terms like `confirm` or `confirm_` to get started.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-24 by thisiskeithb

thinkyhead, 2021-04-25:
Suggestions are always welcome. Such an option may be introduced in the future if someone feels motivated. I'm personally fond of the confirm dialogs because I would frequently start printing the wrong object without realizing it, but I have not made that error at all since I added the dialogs.