MarlinFirmware/Configurations #474: Config Reversing Stepper Direction when adjusting feedrate and mixing stepper

jameswis01, 2021-04-25:
### Bug Description

TMC2209  - BLTouch - sensorless homing.  Works perfect.  Lower feedrates and acceleration .  Works Perfect.  Enable Mixing Steppers.  X AXIS REVERSES!!!!!   Comment Out Mixing Steppers, Recompile and issue still exists...  Tried with the most recent and last 4 versions of marlin... New configs everytime.  Can duplicate EVERYTIME!!!!
### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

Setup - X, Y, Dual Z, Dual Extruders, Mixing Nozzle
TMC2209's Sensorless, No stealth, no hyper. w/ BLTouch.
Configure settings. test.  Lower Feedrates & Acceleration.  X Axis Reverses homing direction.. After enabling Mixing Steppers.. Direction on touchscreen & LCD are still moving correctly.
  Disable mixing steppers, and issue persists.

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**
X axis homes in wrong direction everytime.  Moves in correct direction from LCD and Touchscreen.

**Actual behavior:**
read above.
[Buggy Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/6373431/Buggy.Marlin.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-04-26:
This is not the place to report bugs in anything other than the Configuration file examples.

When submitting a bug report you always have to test on the latest bugfix version marlin.  
Provided config files shows CONFIGURATION_H_VERSION 020006
This is a quite old version of marlin now.
We have had 4 releases since then and bugfix on top of that.

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands.  If the issue should exist the please open an issue on  https://github.com/MarlinFirmware/Marlin/issues/new/choose

jameswis01, 2021-04-26:
Did you read my report?  I reported this issue persisted up through the
latest..  I rolled back trying to find a working version.

thinkyhead, 2021-04-28:
This should be on the Marlin repo. I'll transfer the issue there.