MarlinFirmware/Configurations #476: Configuration version mismatch

festro, 2021-04-29:
### Bug Description

I receive an error compiling marlin stating "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h." and "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h." 

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/6396161/Configuration.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

Compile Marlin following steps in following video "https://www.youtube.com/watch?v=-Gdk0wHg51w&t=344s" with Visual Studio Code

**Expected behavior:**

Compilation should succeed
 
**Actual behavior:**

I receive the aforementioned errors and fails

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-04-29:
You have config files for a newer version of marlin that the code you are running.
Get the correct config files. 

This is not a bug.

ellensp, 2021-04-29:
Provided configs are for bugfix branch of marlin code

ellensp, 2021-04-29:
Most likely you have not change the config branch before downloading configs

```
NB If you are downloading these configuration files in a zip file from the CODE button, you first must select the branch that matches the version of Marlin you are compiling.

Branches
Release: For most Marlin users. Select the version that matches the Marlin version you are compiling.

Bugfix: For users compiling the bugfix branch of Marlin.

Import: For developers maintaining the Configuration examples.
```

closed 2021-04-29 by ellensp

ellensp, 2021-04-29:
Use these configs for marlin 2.0.7.2 https://github.com/MarlinFirmware/Configurations/archive/refs/heads/release-2.0.7.2.zip

ellensp, 2021-04-29:
At this time if you want a BigTreeTech SKR 1.4 Turbo in a ender 3 pro, You have to use bugfix branch code
These configs https://github.com/MarlinFirmware/Configurations/archive/refs/heads/bugfix-2.0.x.zip
With this code https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.0.x.zip

festro, 2021-04-29:
Ok thanks.