MarlinFirmware/Configurations #478: Can we get a config file for the SKR Mini E3 2.0 board on a CR10 S5?

Rhragnarok, 2021-05-02:
### Description

Please make a Config file for the CR10 S5 so that we can use that as a starting point for this combination.

### Configuration Details

I would like a Config file for the CR10 S5 using a SKR Mini E3 2.0 board

* Manufacturer: Creality
* Model: CR10 S5
* Motherboard: SKR Mini E3 2.0 board

#### Additional Information

There is a config file in there for the Creality Silent Board.  But not for the SKR.   I was hoping someone had one for this printer that I could use as a starting point.  I am not an expert on how to set every setting so I am trying to use a CR10S as a starting point but there are lots of differences and I just dont have the knowledge base to figure those out on my own.

ellensp, 2021-05-05:
The CR10 S5 comes with a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. This LCD controller requires 2 * 10 pin plugs the SKR Mini E3 2.0 only has 1 10 pin plug.

You cannot use this LCD on this Conntroller.

ellensp, 2021-05-05:
Carefully try the PR, as noted LCD may not work.

ellensp, 2021-05-05:
This youtube seems to show 2 * 19 pin cables to lcd.

ellensp, 2021-05-05:
looks to be this

[image]

So you are in luck.
If you move EXP1 to EXP3 and unplug EXP2 from the lcd
This lcd is then a compatible with a CR10_STOCKDISPLAY

Rhragnarok, 2021-05-05:
What are you talking about?  My question has nothing to do with the Display controller...

I was asking about Firmware configuration files for the CR10 S5 that have been tested and work reliably so that we dont have to try to make it up and guess what the settings need to be for this printer when using the SKR Mini E3 2.0.

ellensp, 2021-05-05:
Sorry for trying to help.

thisiskeithb, 2021-05-05:
> What are you talking about? My question has nothing to do with the Display controller...

In order to build a config, you need to define the hardware that makes up your 3D printer. This includes knowing what kind of LCD/display you have.

The SKR Mini E3 V2 has a single 10-pin EXP connector which is only compatible with single EXP connector LCDs like those commonly found in many Creality printers (using `CR10_STOCKDISPLAY`). Some Crealtiy models use dual EXP connector displays and need two cables (`REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`). Other displays can be made compatible with a custom wiring harness.

Pop open the control box and see if your LCD has all three EXP ports on the back of the display. This varies from printer to printer/batch to batch/year to year so you may be in luck and can simply connect a single EXP cable between EXP3 on the LCD to the single EXP connector on the SKR Mini E3 V2 board.

Rhragnarok, 2021-05-05:
Yes I understand all of that now.  But that is not my issue or concern with this issue I created.  My only concern is getting a Configuration file for the CR10S5 for the SKR Mini E3 board.

I mean yes this information is important for someone else who might be doing this board on a completely stock CR10S5.  But I did not ask about the screen nor care atm.  I am using the BTT TFT35 for my setup.  So that was all working pretty easily.

I just want that Configuration.h file along with anything else thats needed to create a reliable firmware for that board with that printer.

thisiskeithb, 2021-05-05:
> Yes I understand all of that now. But that is not my issue or concern with this issue I created. 

> I just want that Configuration.h file along with anything else thats needed to create a reliable firmware for that board with that printer.

That’s not how this works.

You don’t just get to demand that we put in all the work to building a proper config *for free* for you and then refuse to answer questions or even attempt to look at your own hardware.

Rhragnarok, 2021-05-05:
Huh?  that's not what I was doing.   When I click "New Issue" on this page it has an option to "Request a Configuration".  I assumed that's for people to request firmware configurations that would be useful to the community.   So I put one in.  Because I have a CR10S5 and I went to try to get it working with this board and it was a big headache because I don't have the knowledge to make it on my own.  So I figured since there is a Config for this printer on the CrealityBoard, and a bunch for a bunch of other boards I thought maybe someone should request one for the SKR board on the S5.

ellensp then pointed out that this board wouldn't work stock with the CR10S5 because of the ribbon wiring issue.  But that had nothing to do with my request so I was confused.  And I tried to clarify my request.

I was not demanding anything.  I was trying to follow the rules of the issues request.

And I am happy to answer questions.  But what type of display controller I happen to have is irrelevant here.  (I am using the TFT35 btw and it has to be set to CR10_STOCKDISPLAY to work)  Since this is merely a request for a CR10S5 firmware configuration for the SKR board.  That means stock so it would be setup by default with the stock display in the config.   If I then used it an happened to use a different display controller I would be on the hook for changing that on my own.

And just to clarify one more time...you have both been focusing on the display...as if that was a problem I was asking for help on.  I never said anything about a display issue, nor has it been a problem for me.  So you had me very confused.

thisiskeithb, 2021-05-05:
You need a board that supports the stock CR-10 S5 configuration with two EXP headers (`REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`).

The SKR Mini E3 V2 only has a single EXP header and is not compatible.

closed 2021-05-05 by thisiskeithb

Rhragnarok, 2021-05-05:
[image]

thisiskeithb, 2021-05-05:
BigTreeTech's readme is incorrect. I'll let them know.

The SKR Mini E3 V2 only has a single EXP header. [From the pins diagram](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf):

[image]

Rhragnarok, 2021-05-05:
Well I can't speak to if it will work with a Stock Display.  But I have it working with a TFT35 display just fine right now.   I ended up using the Cheetah 5.0 firmware and so far I was able to do a simple basic print with it with no issues.