MarlinFirmware/Configurations #480: [FR] Anycubic kossel linear plus (#define Z_PROBE_FEEDRATE_FAST)

pasikov123, 2021-05-03:
When trying to compile and load the marlin 2.0.7.2 build for Anycubic kossel linear plus with a btt skr 1.4 board using visual studio code, we get the error  Marlin\src\inc/../../Configuration.h:1213:31: error: 'homing_feedrate_mm_m' was not declared in this scope

My configurations
sanicubic kassel linear plus
marlin 2.0.8
MOTHERBOARD BOARD_BTT_SCR_V1_4
#define ANYCUBIC_PROBE_VERSION 2

Steps to play back
1-Downloaded marlin version 2.0.8
2-Settings in Configuration. h
3-Build

Expected behavior: successful build/upload

Additional information
It is not clear what value to specify in Z_PROBE_FEEDRATE_FAST ? In version 2.0.7.2, there was a value of (30*60). What is the HOMING_FEEDRATE_MM_M variable used for?

ellensp, 2021-05-03:
The error is due to using new 2.0.8 Configs on 2.0.7.2 code.
This will never work, 
You need to make sure the code and the configurations versions match
Upgrade your marlin to 2.0.8

closed 2021-05-03 by ellensp

pasikov123, 2021-05-06:
I have marlin Marlin-2.0.8 in the configuration for Kossel LinearPlus
In the configuration, the value goes
## define Z_PROBE_FEED RATE_FAST homing_feedrate_mm_m. z
Because of homing_feedrate_mm_m. z does not collect the bin file

ellensp, 2021-05-06:
a PR has been created to fix the Kossel Linear Plus configs. 
https://github.com/MarlinFirmware/Configurations/pull/485