MarlinFirmware/Configurations #487: [BUG] Creality/Ender-3 Pro/BigTreeTech SKR 1.4 Turbo E-steps and RMS set wrong

JoeDaStudd, 2021-05-06:
### Bug Description

E steps set incorrectly for stock Ender 3 Pro and the TMC RMS is too low.
The E-steps causes nearly double the extrusion and the RMS causes very noticeable skips on the extruder, potentially issues with X, Y and Z, but I corrected this before fully testing.

Potentially the combination would mean a newbie wouldn't fully notice the issues as they partially counter each other. 
[BigTreeTech SKR 1.4 Turbo.zip](https://github.com/MarlinFirmware/Configurations/files/6436190/BigTreeTech.SKR.1.4.Turbo.zip)

### Configuration Files

Issues in Configuration.h
`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 141 }`
Should be 
`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 93}`

Issues in Configuration.h
```
#define X_CURRENT       580
#define Y_CURRENT       580
#define Z_CURRENT       580
#define E0_CURRENT      650
```
Should be or close to 
```
#define X_CURRENT       760
#define Y_CURRENT       760
#define Z_CURRENT       760
#define E0_CURRENT      900
```

### Steps to Reproduce

Update SKR 1.4 firmware using the config files, print any calibration print and notice clicking on the extruder stepper and over extrusion in the print

**Expected behavior:**

Normal printing with minimal calibration 

**Actual behavior:**

Over extrusion and step skipping

#### Additional Information

Either the config file needs adjusting or the readme needs to make note it's setup for a non stock extruder setup and notes to relevant fields to change.

thisiskeithb, 2021-05-06:
> Either the config file needs adjusting or the readme needs to make note it's setup for a non stock extruder setup and notes to relevant fields to change.

Either way since that config is specific to one particular user and they even noted _"I may have some other options enabled that I don't remember so be careful to check everything that corresponds to your machine._"

Since you've provided a fix, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

ellensp, 2021-05-07:
I agree on e-steps, but E0_CURRENT 900 is higher than any other examples configurations for this machine.

thisiskeithb, 2021-05-07:
> I agree on e-steps, but E0_CURRENT 900 is higher than any other examples configurations for this machine.

That and `*_CURRENT` settings match stock motor/config settings, so increasing them here doesn’t make sense.

JoeDaStudd, 2021-05-07:
I tried increasing the current using to M906 and even at 800 I was getting the odd skip on the extruder.
After some looking around (I didn't want to increase it further without checking) I came across this [thread](https://www.reddit.com/r/BIGTREETECH/comments/ct5j8k/stepper_driver_current_settings/) on the Reddit BTT sub about current settings for the TMC2209.

After changing it to 900 for E0 all the skipping has gone and after a 3-4 hour print the stepper was more then cool enough to touch.
It could be that it's a fresh roll of 1kg filament  and the stock holder needing more torque.

The extruder stepper motor is rated about 20% higher then the X, Y and Z motors on the Ender 3 which is why there should be a variation between them, the marlin default 800 is across all drivers and assumes they are the same model.

I've just remembered there are at least two other bugs/issues in this config files I found and amended before my initial compile.

Configuration.h
```
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
```
Should be 
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
```
This is taken from the BTT config file for the 1.4 Turbo on [their github page](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Firmware/Marlin-2.0.x-SKR-V1.4-Turbo/Marlin/Configuration.h)

Configuration_adv.h
`#define HOMING_BUMP_MM      { 5, 5, 2 } `
Should be 
`#define HOMING_BUMP_MM      { 0, 0, 2 } `

This config is for the TMC2209 with sensorless homing on X and Y and Marlins documents recommends zeroing HOMING_BUMP_MM on axis with sensorless homing for best results.

I really appreciate people providing the example config files, but if I didn't have an old copy of the firmware settings for the SKR 1.4 Turbo with 2209's I made for another printer I'd have been scratching me head a lot.

SamppaD78, 2021-05-24:
> ### Bug Description
> E steps set incorrectly for stock Ender 3 Pro and the TMC RMS is too low.
> The E-steps causes nearly double the extrusion and the RMS causes very noticeable skips on the extruder, [...]

Im using Ender 3 Pro with  SKR mini E3 2.0 TMC 2209 . Im running these settings without any issues what so ever. All motors running nice and cool 40-50C
No skipping or under extrusion
X_CURRENT       500
Y_CURRENT       500
Z_CURRENT       650.. dual Z setup 
E0_CURRENT     600 dual gear extruder E-steps 144

So first check printer settings,extruder gears  and hotend .All examples provided in marlin needs to be checked to make sure that they are suitable for your machine.
E-steps of 93 are for stock Creality extruder.

SamppaD78, 2021-05-24:
To add to my previous post is that config files current settings X580,Y580,Z580,E650 are as they supposed to be.
If you run motors on  0.76A they will get hot. Bigtreetech  had firmware  issue where the current for tmc drivers was set to high to 0,8A and eNDER printer motors got extremely hot as they are 0.8A motors.  After this  mistake i compile firmware myself :-)

thinkyhead, 2021-08-22:
Are these changes still recommended and do we all agree? The home bump mm change makes good sense to me.
```cpp
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define HOMING_BUMP_MM { 0, 0, 2 }
```

thisiskeithb, 2021-08-22:
> Are these changes still recommended and do we all agree?

Looks good to me.

closed 2021-09-09 by thinkyhead