MarlinFirmware/Configurations #492: [BUG] (Meshbed edit shows one row less. Is saved one line wrong.)

Stein-Wolf, 2021-05-16:
### Bug Description

Mesh Bed Edit shows 1 row less than leveled and if you level e.g. point 2 in second row, it is saved as point 2 in row 1.

More accurate: If you have a grid of 5x5 on MBL you only get shown a grid of 5x4 in the edit section. The nozzle cannot go to the last row of the bed but also save the wrong position.

If you edit e.g. the point marked as "o" (5th "y" row isn't shown and you cannot goto with the nozzle):
5: y-y-y-y-y
4: x-x-x-x-x
3: x-x-x-x-x
2: x-x-x-o-x
1: x-x-x-x-x
it will be saved as this point marked as ("o"):
5: y-y-y-y-y
4: x-x-x-x-x
3: x-x-x-x-x
2: x-x-x-x-x
1: x-x-x-o-x

### Configuration Files
[conf.zip](https://github.com/MarlinFirmware/Configurations/files/6489219/conf.zip)
Original is from "digant" at https://www.thingiverse.com/thing:4294049/ - version for Sidewinder X1. I think originated from bugfix version.
I haven't done changes except extruder calibration. Digant, as far as I can see, hasn't done anything except enable and disable sections and do some calibrations for the sidewinder X1.

### Steps to Reproduce

1. Make MBL (manuel) (grid 5x5)
2. Than goto Mesh Edit in the LCD. See that it only shows a grid of 5x4.
3. Anyway edit a point e.g. second row (after seeing first layer is of). Z height e.g. is "-0.08" - new z height is e.g. "0.13"
4. Save to EProm
5. Do Mesh Edit again
6. Goto the same point - notice that heigt is still "-0.08" but point below second row now has heigt "0.13".

**Expected behavior:**

Show grid 5x5 is MBL is 5x5. I think this will maybe fix the wrong point saving aswell.

**Actual behavior:**

See above.

#### Additional Information

See thingiverse link above. First picture shows the 5x5 grid as it should be.
See https://www.youtube.com/watch?v=bAXEwa6roL0&lc=UgzDAM0s2m70-TkLatR4AaABAg.9MCaEs7mNwz9MgWeaGqCnf
There you can see the wrong grid 5x4.

I hope my first post is ok. This is all new for me.

ellensp, 2021-05-16:
@thinkyhead please move to Marlin issues

Stein-Wolf, 2021-05-16:
@ellensp What you mean? I'm not thinkyhead and I think I'm in issues. 
Am I not?

ellensp, 2021-05-17:
You posted this in MarlinFirmware/Configurations issues. Which if for issues with the provided example configuration files.
It needs to be in MarlinFirmware/Marlin issues. Which is for issues in the main marlin code.
Thinkyhead is the main marlin developer.  
Preceding comment with a @ mean this comment is directed at that user.

Sonicboom247, 2021-05-17:
@Stein-Wolf 
I am working on a custom firmware for the Sidewinder X1, i have it about 98% complete and am just waiting on the 5x5 mesh editing, will test tonight with a different configurations like 6 rows to see if it can get 5.

Stein-Wolf, 2021-05-17:
@ellensp - Sorry. I was so focused on getting an answer, that I didn't realise that thinkyhead should move this to the right place. 
@Sonicboom247 - Yeah!

Stein-Wolf, 2021-05-17:
So, now I get a new information that the original developer "digant" has released an update at thingverse, which probably fix the problem. Maybe you can see what he did.
I cannot test this until weekend, but I guess the problem is fixed for me. 
Should I close this bug report here, or would you implement/test/repair this original firmware first?
@Sonicboom247 - I'm anyway still interested in you firmware!

thisiskeithb, 2021-05-17:
We can’t support forks or random copies of Marlin, so please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Sonicboom247, 2021-05-17:
@thisiskeithb Thank you!, i am testing with build 2.0.8 and so far loving it
@Stein-Wolf Thanks for the update!, I will still produce my firmware as it has much nicer icons as well as taking advantage of more of the great options Marlin offers

Sonicboom247, 2021-05-18:
> We can’t support forks or random copies of Marlin, so please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

@thisiskeithb can you inform me of the proper version path i should be testing with? i am currently testing with 2.0.8.1 with the assumption it was updated with everything from bugfix-2.0.x? should i always stay on bugfix or when can i move to a stable version?  thank you !

thisiskeithb, 2021-05-18:
> can you inform me of the proper version path i should be testing with?

Now that `2.0.8` & `2.0.8.1` have been released, they will be close to the latest `bugfix-2.0.x`, but there were comments here about using forks/other copies of Marlin and we have no control of what was done in those copies.

I just wanted to make sure everyone reporting issues were using [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) from this repo so we're all working from the same code.

Sonicboom247, 2021-05-18:
Ok excellent, wanted to make sure I was not misguiding any available support. That being said, I did test the printer_event_leds and am not able to get them to switch from blue to violet and so on. Although I am not able to locate where exactly the color changes are defined.