MarlinFirmware/Configurations #494: [BUG] BLTouch/Z Axis homing working with 2.0.2 Waggster mod but not with 2.0.8.1

jcastro, 2021-05-18:
### Bug Description

I have an Artillery genius printer to which I've added the BL Touch 3.1 sensor. Currently it's working with any problems using Waggster mod BLTouch v1.5 firmware, which is based on 2.0.2

However I'm trying to update the printer to marling 2.0.8.1. I've tried to move all parameters from Configuration.h. file to Marlin 2.0.8 using as a base the Artillery Genius configuration example at https://github.com/.../config/examples/Artillery/Genius but I'm unable to make the homing (G28) to work correctly. I can't find which parameter or configuration is failing or missing... or it's a bug

Below there's the source code of what I'm using, both from Waggster mod and from Marlin 2.0.8 (which is the one I'm trying to configure myself). I'm also adding videos to show you what happens when I send the G28 command to the printer

Waggster mod firmware: https://d.pr/f/F0gfYS
Marlin 2.0.8: https://d.pr/f/eMOztd
Video Waggster mod, working ok: https://d.pr/v/A35WW4
Video Marlin 2.0.8: https://d.pr/v/VGBaP6 (as you can see, it never goes down to do the measuring)
I've also tried Marlin 2.0.6, latest marlin bugfix and import-2.0.x branch... same thing

### Configuration Files

Waggster mod firmware: https://d.pr/f/F0gfYS
Marlin 2.0.8: https://d.pr/f/eMOztd

### Steps to Reproduce

1. Upload the firmware
2. Send G28 command
3. At the moment of doing the Z axis it will stay up 

**Expected behavior:**

After sending G28 it should home all axis, but not working for Z

**Actual behavior:**

With the Waggster mod based on marlin 2.0.2 it's wokring correctly, with Marlin 2.0.8.1 and some other versions I mentioned above I have this problem where when sending G28 command it won't correctly do the Z axis (you can check the video for better reference, these shows what happens when I send the G28)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-05-18:
In the Waggster they have modified Marlin/src/pins/ramps/pins_RAMPS.h
they have swapped  the defaults 
```
    #define Z_MIN_PIN      18
    #define Z_MAX_PIN      19
```
to
```
    #define Z_MIN_PIN      19
    #define Z_MAX_PIN      18
```
So my guess would be that your bltouch is plugged into the Z_MAX plug on the controller and this is what they used to swap it over.

You can just add this to your Configuration.h 
```
#define Z_MIN_PIN      19
#define Z_MAX_PIN      18

``` 
( pins_RAMPS.h is written so if these are defined already it doesn't change them, so you don't need to modify pins_RAMPS)

ellensp, 2021-05-18:
What you really should do is move the probe to Z_min or set #define Z_MIN_PROBE_PIN 19

ellensp, 2021-05-18:
Updated:
What you really should do is move the probe to Z_min or set
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING
#define Z_MIN_PROBE_PIN 19

closed 2021-05-18 by ellensp

jcastro, 2021-05-18:
@ellensp thank you so much, I will give this a try. I'm a total noob on this... I would have never find out without your help. Thanks!