MarlinFirmware/Configurations #495: Power Loss Z-Raise not raising

Sutan24, 2021-05-18:
Hi there, Well Marlin 2.0.8 having a problem regards to the power loss recovery matter.
During Power loss:
1.	POWER_LOSS_ZRAISE 5: Z-axis not raising (nothing happening when power loss)
2.	POWER_LOSS_RETRACT_LEN 10: is not retracting (nothing happening when power loss)
When power resumes:
X & Y axis moves filament purge working fine as assigned, continue printing
My controller board : BIGTREE_SKR_PRO 1.1
Printing method: SD
Slicer: Cura
I have a custom-built UPS 12V that can be programmed to make it LOW or HIGH power loss signals when a power failure occurs. Has a precision intelligence that signals power loss. Was working absolutely fine with 2.0.
I spend a lot of time changing both HIGH and LOW power loss states. And tried Pullup, Pulldown, and un command both. Assigned different power loss pins.
Power loss recovery works like there is no pin assigned.
I measured the voltages when changing pullup/down/un commended. It’s working accordingly.
My guess is the power loss pin not working with the firmware.
Also tried to enabled #define POWER_LOSS_RECOVER_ZHOME when compile msg throwing an error: cannot be used with Z safe Homing
Moreover tried #define PLR_ENABLED_DEFAULT tried both false & true
Also, tried sending commands manually M413 S1 & M413 S0 with no luck at all.
Any help would be appreciated.
[Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/6504822/Marlin.zip)

thisiskeithb, 2021-05-18:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/21918

thisiskeithb, 2021-05-18:
Please don’t create multiple issues across multiple repos. Someone will get to your issue when they have time.

closed 2021-05-18 by thisiskeithb

virus-sss, 2023-01-15:
на MKS ROBIN PRO (IDEX ) такая же проблема