MarlinFirmware/Configurations #502: [BUG] (short description)

fumanschuh, 2021-05-24:
Sapphire Plus MKS UI 
When i define the MKS UI the touchscreen is disabled.

ellensp, 2021-05-29:
as fas as I am aware the MKS UI is only for TFT's with an encoder. 

eg [image]

thisiskeithb, 2021-05-29:
> as fas as I am aware the MKS UI is only for TFT's with an encoder.

No, it works on MKS' other TFTs as well.

thisiskeithb, 2021-05-29:
Interestingly, the author of the config specifically disables `TOUCH_SCREEN` if you enable the special/config-specific flag `SAPPHIRE_PLUS_MKS_UI`, so it's intentional:

https://github.com/MarlinFirmware/Configurations/blob/dd4de330913ae7e416a66a33e4b76a5fe8ece807/config/examples/Two%20Trees/Sapphire%20Plus/Sapphire%20Plus%20V2/Configuration.h#L2609-L2611

Outside of this printer's specific config, MKS's UI works fine with `MKS_ROBIN_TFT35`.

thinkyhead, 2021-06-12:
Fill out a bug report at the main project.

closed 2021-06-12 by thinkyhead