MarlinFirmware/Configurations #506: [FR] Configuration Update

choppman, 2021-05-30:
### Description

https://github.com/MarlinFirmware/Configurations/blob/release-2.0.8.2/config/examples/Creality/Ender-5%20Pro/BigTreeTech%20SKR%20E3%20Turbo/Configuration_adv.h

It has:
   #define SHORT_MANUAL_Z_MOVE 0.025 // (mm) Smallest manual Z move (< 0.1mm)

It should be:
  #define FINE_MANUAL_MOVE 0.025    // (mm) Smallest manual move (< 0.1mm) applying to Z on most machines

* Manufacturer: BTT
* Model: Ender 5 Pro
* Motherboard:  BTT SKR E3 Turbo

#### Additional Information
it could be like this in the other configuration files, but without making this change it fails the sanity check.

ellensp, 2021-06-06:
Confirmed
4 Config files still have this old keyword
config/examples/Creality/Ender-5 Pro/BigTreeTech SKR E3 Turbo with BLTouch/Configuration_adv.h
config/examples/Creality/Ender-5 Pro/BigTreeTech SKR E3 Turbo/Configuration_adv.h
config/examples/Creality/Ender-5/BigTreeTech SKR E3 Turbo with BLTouch/Configuration_adv.h
config/examples/Creality/Ender-5/BigTreeTech SKR E3 Turbo/Configuration_adv.h

ellensp, 2021-06-06:
Pr created, this will be added soon

closed 2021-06-06 by ellensp