MarlinFirmware/Configurations #513: Ender 5 BTT SKR 1.4 Turbo config wont compile

brysoncotton, 2021-06-06:
### Bug Description

When I go to compile the ender 5 code with the BigTreeTech SKR 1.4 Turbo board config, the Auto Build Marlin terminal in VS Code displays an error. I tried both the 2.0.x and the Bigfix-2.0.x but both came back with this error

Verbose mode can be enabled via `-v, --verbose` option
arm-none-eabi-g++.exe: fatal error: cannot execute 'cc1plus': CreateProcess: No such file or directory
compilation terminated.
CalledProcessError: Command '"C:\Users\Bryson\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe" -D__MARLIN_FIRMWARE__ -DNDEBUG -DU8G_HAL_LINKS -D__MARLIN_DEPS__ -w -dM -E 
-x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\Bryson\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175:
    env.SConscript(item, exports="env")
  File "C:\Users\Bryson\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Bryson\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Bryson\Desktop\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 309:
    apply_features_config()
  File "C:\Users\Bryson\Desktop\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 136:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\Bryson\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
    load_marlin_features()
  File "C:\Users\Bryson\Desktop\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 273:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\Bryson\.platformio\python3\lib\subprocess.py", line 424:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\Bryson\.platformio\python3\lib\subprocess.py", line 528:
    raise CalledProcessError(retcode, process.args,
=================================================================================== [FAILED] Took 0.56 seconds ===================================================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:00.556
============================================================================= 1 failed, 0 succeeded in 00:00:00.556 ============================================================================= 
PS C:\Users\Bryson\Desktop\Marlin-2.0.x> echo "done" >C:\Users\Bryson\AppData\Local\Temp\ipc
PS C:\Users\Bryson\Desktop\Marlin-2.0.x> 

### Configuration Files

[Ender 5 Config..zip](https://github.com/MarlinFirmware/Configurations/files/6604342/Ender.5.Config.zip)

I have not made any modifications to any lines in the code.

### Steps to Reproduce

1. Attempt to compile marlin with the BTT SKR 1.4 Turbo mainboard config

**Expected behavior:**

I expected the file to compile easily

**Actual behavior:**

Wont compile displays an error

#### Additional Information

[image]

ellensp, 2021-06-06:
quit vscode. Delete hidden directory C:\Users\Bryson\.platformio  Restart vscode and try again

brysoncotton, 2021-06-06:
That worked, thank you so much!!!

closed 2021-06-06 by brysoncotton