MarlinFirmware/Configurations #521: configuration request

chappy1978, 2021-06-13:
Can you please add a config for a Creality CR10S Pro V2?
Thank You

### Description

### Configuration Details

* Manufacturer:
* Model:
* Motherboard:

#### Additional Information

* Provide links to the requested printer/configuration.

thinkyhead, 2021-06-13:
Someone with the hardware will have to get it sorted out. If you manage to produce working configs, please submit!

LTSpenny, 2021-06-22:
@thinkyhead new to GitHub how do we submit configs?

LTSpenny, 2021-06-22:
@thinkyhead belay my last. Just found info on how to do so.

burnsy180, 2021-08-09:
Has there been any progress on this configuration please? @LTSpenny have you got a working config you can share?

SushiKiIIer, 2021-11-18:
I need a set of config files as well; or a link to a guide for setting one up from scratch and ill give it a try xD. Doesn't look like the forum above is actually filled out so in case it helps someone get the config file sorted ill do it. Sorry if the links aren't to technical pages for the boards  I haven't been able to find much so far info wise.

thanks again

       **Description:** CR-10S Pro V2 Stock configuration files ( .h & adv.h )

**Configuration details:**
          - Manufacturer: Creality
          - Model: CR-10S Pro V2
          - Motherboard: Creality V2.4.1

**Additional Information**
          - Printer Link: CR-10SProV2: https://tiny-machines-3d.myshopify.com/products/cr10sprov2?_pos=3&_sid=74ca44ec2&_ss=r
          - Motherboard Link: Creality V2.4.1: https://tiny-machines-3d.myshopify.com/products/creality-10s-pro-motherboard

thinkyhead, 2022-01-26:
Duplicate of #649

closed 2022-01-26 by thinkyhead