MarlinFirmware/Configurations #522: ender 3 max w/BL touch request

endl, 2021-06-13:
would really like a working config for this set up and/or one with a SKR mini e3 v2 as the main board

thinkyhead, 2021-06-13:
Someone with the hardware will have to tackle the problem. If you figure it out please submit your results.

LTSpenny, 2021-06-21:
I can make one. I have one configured for the SKR E3 Turbo that i can make changes to suit yours. How do i add a config example to the repository?

endl, 2021-06-21:
No idea I'm not even sure half my settings are correct or working properly

LTSpenny, 2021-06-21:
@endl well i can make one and just link it for you. What board do you want it for. Skr mini e3?

endl, 2021-06-21:
Yea skr mini e3 v2

LTSpenny, 2021-06-22:
https://drive.google.com/drive/folders/1blPp7cNIaFi7SdDJ0D_c9sSqE35h3H9x?usp=sharing

try this configuration, its mostly what I'm running on my max. Hope that gives you a good starting point, make sure you compile in the right environment. Also this was a config file from the last bug build a few days ago, did not check if there's been an update on that. Cheers.

LTSpenny, 2021-08-23:
Please close ticket if this solved your problems.

closed 2021-08-23 by endl