MarlinFirmware/Configurations #525: [BUG] Unable to compile unedited config  dual z axis

Octoclops, 2021-06-14:
### Bug Description

When trying to compile using PlatformIO the compiler throws a few errors:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1164:19: error: #if with no expression
 1164 |   #if IS_ULTIPANEL
      |                   ^
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:55,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/drivers.h:80:51: error: missing binary operator before token "("
   80 | #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
      |                                                   ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/drivers.h:119:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
  119 | #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) || AXIS_DRIVER_TYPE(A,TMC2160) \
      |                               ^~~~~~~~~~~~~~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2616:7: note: in expansion of macro 'AXIS_IS_TMC'
 2616 |   #if AXIS_IS_TMC(I)
      |       ^~~~~~~~~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/drivers.h:80:51: error: missing binary operator before token "("
   80 | #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
      |                                                   ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/drivers.h:119:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
  119 | #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) || AXIS_DRIVER_TYPE(A,TMC2160) \
      |                               ^~~~~~~~~~~~~~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2625:7: note: in expansion of macro 'AXIS_IS_TMC'
 2625 |   #if AXIS_IS_TMC(J)
      |       ^~~~~~~~~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/drivers.h:80:51: error: missing binary operator before token "("
   80 | #define AXIS_DRIVER_TYPE(A,T) AXIS_DRIVER_TYPE_##A(T)
      |                                                   ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../core/drivers.h:119:31: note: in expansion of macro 'AXIS_DRIVER_TYPE'
  119 | #define AXIS_IS_TMC(A)   (    AXIS_DRIVER_TYPE(A,TMC2130) || AXIS_DRIVER_TYPE(A,TMC2160) \
      |                               ^~~~~~~~~~~~~~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2634:7: note: in expansion of macro 'AXIS_IS_TMC'
 2634 |   #if AXIS_IS_TMC(K)
      |       ^~~~~~~~~~~

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20E3%20Turbo/Dual%20Z

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Add config files to Marlin folder
2. Click the check in PlatformIO to compile

**Expected behavior:**

The firmware should compile

**Actual behavior:**

The compiler throws errors and doesn't compile

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-06-15:
You should not be using import branch.

From the readme.md of the Configurations repository

NB If you are downloading these configuration files in a zip file from the CODE button, you first must select the branch that matches the version of Marlin you are compiling.

Branches
Release: For most Marlin users. Select the version that matches the Marlin version you are compiling.

Bugfix: For users compiling the bugfix branch of Marlin.

Import: For developers maintaining the Configuration examples.

ellensp, 2021-06-15:
I tested bugfix configuration files  on bugfix code and It compiles without issues.

closed 2021-06-15 by ellensp