MarlinFirmware/Configurations #535: [FR] Creality Ender 3 V2 with SKR Mini E3 V2.0

blazewicz, 2021-07-01:
### Description

The [BIGTREETECH SKR Mini E3 V2.0](https://www.biqu.equipment/products/bigtreetech-skr-mini-e3-v2-0-32-bit-control-board-integrated-tmc2209-uart-for-ender-3?variant=31995437744226) board is viable drop-in replacement for Creality's v4.2.2 board. Mine got fried (bed heater mosfert blew up) and SKR seemed to be the best alternative to buying new board from Creality.

Unlike previous versions of Ender 3 ([1], [2]), the V2 doesn't have upstream configurations for this board. It was already discussed in https://github.com/MarlinFirmware/Marlin/issues/19858, but at that point adding support for DWIN display wasn't possible without modifications to Marlin's code.

One culprit is that SKR display port has slightly different pinout than DWIN and requires slight modification to the ribbon cable. This would require additional instructions in configuration's directory.

Full documentation can be found in [BIGTREETECH's GitHub](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3), there is active discussion with success stories here:

* https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/424

### Configuration Details

* Manufacturer: Creality
* Model: Ender 3 V2
* Motherboard: SKR Mini E3 V2.0

[1]: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0
[2]: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%202.0

blazewicz, 2021-07-02:
Installation is very easy and board has exactly same dimmensions as Creality v4.2.2 and v4.2.7, connector layout is also very similar.

Only hardware modification required is a display cable adapter made with 8 dupont F-M cables.

[image]

below are all required modifications to Ender 3 V2 configurations (thanks @jernejp21, https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/424#issuecomment-707094814):

```diff
--- ../Configurations/config/examples/Creality/Ender-3 V2/Configuration.h	2021-07-02 10:58:20.000000000 +0200
+++ Marlin/Configuration.h	2021-07-02 12:38:43.000000000 +0200
@@ -104,7 +104,7 @@
  *
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-#define SERIAL_PORT 1
+#define SERIAL_PORT -1

 /**
  * Serial Port Baud Rate
@@ -141,9 +141,17 @@

 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_CREALITY_V4
+  #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0
 #endif

+#define BTN_ENC                 PB5
+#define BTN_EN1                 PB15
+#define BTN_EN2                 PB8
+#define BEEPER_PIN              PA15
+#define ENCODER_PULSES_PER_STEP 4
+#define LCD_SERIAL_PORT         1
+#undef SPEAKER
+
 // Name displayed in the LCD "Ready" message and Info menu
 #define CUSTOM_MACHINE_NAME "Ender-3 V2"

@@ -833,9 +841,9 @@
  *          TMC5130, TMC5130_STANDALONE, TMC5160, TMC5160_STANDALONE
  * :['A4988', 'A5984', 'DRV8825', 'LV8729', 'L6470', 'L6474', 'POWERSTEP01', 'TB6560', 'TB6600', 'TMC2100', 'TMC2130', 'TMC2130_STANDALONE', 'TMC2160', 'TMC2160_STANDALONE', 'TMC2208', 'TMC2208_STANDALONE', 'TMC2209', 'TMC2209_STANDALONE', 'TMC26X', 'TMC26X_STANDALONE', 'TMC2660', 'TMC2660_STANDALONE', 'TMC5130', 'TMC5130_STANDALONE', 'TMC5160', 'TMC5160_STANDALONE']
  */
-#define X_DRIVER_TYPE TMC2208_STANDALONE
-#define Y_DRIVER_TYPE TMC2208_STANDALONE
-#define Z_DRIVER_TYPE TMC2208_STANDALONE
+#define X_DRIVER_TYPE TMC2209
+#define Y_DRIVER_TYPE TMC2209
+#define Z_DRIVER_TYPE TMC2209
 //#define X2_DRIVER_TYPE A4988
 //#define Y2_DRIVER_TYPE A4988
 //#define Z2_DRIVER_TYPE A4988
@@ -844,7 +852,7 @@
 //#define I_DRIVER_TYPE  A4988
 //#define J_DRIVER_TYPE  A4988
 //#define K_DRIVER_TYPE  A4988
-#define E0_DRIVER_TYPE TMC2208_STANDALONE
+#define E0_DRIVER_TYPE TMC2209
 //#define E1_DRIVER_TYPE A4988
 //#define E2_DRIVER_TYPE A4988
 //#define E3_DRIVER_TYPE A4988
@@ -1293,9 +1301,9 @@
 // @section machine

 // Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
-#define INVERT_X_DIR false
-#define INVERT_Y_DIR false
-#define INVERT_Z_DIR true
+#define INVERT_X_DIR true
+#define INVERT_Y_DIR true
[... 68 lines cut ...]
+  #define  X_SLAVE_ADDRESS 0
+  #define  Y_SLAVE_ADDRESS 2
+  #define  Z_SLAVE_ADDRESS 1
   //#define X2_SLAVE_ADDRESS 0
   //#define Y2_SLAVE_ADDRESS 0
   //#define Z2_SLAVE_ADDRESS 0
@@ -2769,7 +2769,7 @@
   //#define  I_SLAVE_ADDRESS 0
   //#define  J_SLAVE_ADDRESS 0
   //#define  K_SLAVE_ADDRESS 0
-  //#define E0_SLAVE_ADDRESS 0
+  #define E0_SLAVE_ADDRESS 3
   //#define E1_SLAVE_ADDRESS 0
   //#define E2_SLAVE_ADDRESS 0
   //#define E3_SLAVE_ADDRESS 0
@@ -2813,7 +2813,7 @@
    * Define your own with:
    * { <off_time[1..15]>, <hysteresis_end[-3..12]>, hysteresis_start[1..8] }
    */
-  #define CHOPPER_TIMING CHOPPER_DEFAULT_12V        // All axes (override below)
+  #define CHOPPER_TIMING CHOPPER_DEFAULT_24V        // All axes (override below)
   //#define CHOPPER_TIMING_X  CHOPPER_TIMING        // For X Axes (override below)
   //#define CHOPPER_TIMING_X2 CHOPPER_TIMING_X
   //#define CHOPPER_TIMING_Y  CHOPPER_TIMING        // For Y Axes (override below)
```

blazewicz, 2021-07-02:
Here's Proof-Of-Concept 3DBenchy:

[image]

[image]

blazewicz, 2021-07-05:
Env used in BTT repo, i.e. `STM32F103RC_btt_maple` ([link](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/3419f21a799b2598ec58ea037f4093ddcbbc318f/firmware/V2.0/Marlin-2.0.8.2.x-SKR-mini-E3-V2.0/platformio.ini#L16))  doesn't work, neither does `STM32F103RE_btt_maple`.

With `STM32F103RE_btt` it works, but without USB CDC. Everything works only with `STM32F103RE_btt_USB`.

blazewicz, 2021-07-05:
At the moment enabling serial port simultaneously on USB (serial port -1) and TFT pins (serial port 2) is impossible due to: https://github.com/MarlinFirmware/Marlin/issues/22299

looxonline, 2021-07-07:
> Env used in BTT repo, i.e. `STM32F103RC_btt_maple` ([link](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/3419f21a799b2598ec58ea037f4093ddcbbc318f/firmware/V2.0/Marlin-2.0.8.2.x-SKR-mini-E3-V2.0/platformio.ini#L16)) compiles, but neither display nor serial over USB works.
> 
> With `STM32F103RC_btt` it works, but without USB CDC. Everything works only with `STM32F103RC_btt_USB`.

You can get serial to work by editing the STM32Duino code and changing the USB_EP to single buffer instead of double buffer.

RipperDrone, 2021-07-14:
@blazewicz Hello, I'm tgrying to repcace my Creality 4.2.2 mainboard in Ender 3 V2 with BTT E3 V2 including TFT35 display. On top of that, there is BLTouch hardware and a Hemera direct drive extruder installed. How should I proceed with getting the Marlin firmware right? Have settings working nicely just now, worried how to change them over the easiest way for new mainboard and display...

RipperDrone, 2021-07-18:
@blazewicz Did u test this combo long enough to tell for sure it does not create freeze issues on the printer anymore? Have it up and running now, but getting random freezes... :-( (have created a new issue ticket with more description today. I know this has been a subject in the past already, but latest fixes in serial comm should have fixed it, but apprently haven't for me...)

blazewicz, 2021-07-18:
@RipperDrone I'm not sure if this is the same issue you're refering to, but every few hours USB serial connection resets and so I can't run longer prints from Octoprint. It might be related to USB serial buffer, but I haven't verified it yet.

Also once in a while I get random Z offset error. Once it prints perfect and at next print, after homing nozzle is too high and I get no adhesion. Only full printer restart helps. I use standard Z stop, no ABL.

blazewicz, 2021-07-18:
@RipperDrone have you checked whether your board has original ST chip or GigaDevice clone?

RipperDrone, 2021-07-18:
@blazewicz Thank you for following up.

- no, it's not like an occasional USB buffer overflow every couple hours, it freezes after few minutes already when the printer is warm. After cold start, I MIGHT be able to finish one xyz test cube, but printer will hang on the second already. After that it will hang on every single print, from the point it's starting - sometimes during heating phase already. This, it might be related to so e thermal issue / not necessarily to USB? It happens as well when disconnecting PC from USB and printing from SD card. Of course, the 2 USB ports are opened by Malin, anyways...
- I have not checked for the chip version, but ordered from BTT store directly on Amazon since I was afraid of clones. Cannot check anymore, sending it back and will try a replacement one next week, just to be sure that hardware is ok
- I read that some boards might have flawed 512k flash. Emory in upper half, and I compiled in _btt_512k eng so far. Should I rather try _btt eng instead when firmware filesize fits in?

blazewicz, 2021-07-18:
> no, it's not like an occasional USB buffer overflow every couple hours, it freezes after few minutes already when the printer is warm. After cold start, I MIGHT be able to finish one xyz test cube, [...]

Oh, ok, I was able to successfully run 10h+ prints from SD card so I guess its not related to the firmware, maybe your hardware is faulty.

> I have not checked for the chip version, but ordered from BTT store directly on Amazon since I was afraid of clones. Cannot check anymore, sending it back and will try a replacement one next week, just to be sure that hardware is ok

Sadly now, during the chip shortage its a lottery what you get.

> I read that some boards might have flawed 512k flash. Emory in upper half, and I compiled in _btt_512k eng so far. Should I rather try _btt eng instead when firmware filesize fits in?

I don't think it matters, nothing in linker script is aligned to the end of flash, so as long as binary is <256k there's no difference. Of course its better when linker knows actual size of target flash.

BTW I think `_btt_512k` env was replaced with `STM32F103RE_btt`, you should update your repo.

blazewicz, 2021-07-18:
> You can get serial to work by editing the STM32Duino code and changing the USB_EP to single buffer instead of double buffer.

@looxonline: true, I changed following and USB serial works. I wish there was a switch.

```diff
--- ~/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb/usbd_ep_conf.c	2021-07-19 00:26:39.000000000 +0200
+++ ~/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb/usbd_ep_conf.c	2021-07-19 00:26:30.000000000 +0200
@@ -37,7 +37,7 @@
 #else
   {0x00,       PMA_EP0_OUT_ADDR, PCD_SNG_BUF},
   {0x80,       PMA_EP0_IN_ADDR,  PCD_SNG_BUF},
-  {CDC_OUT_EP, PMA_CDC_OUT_ADDR, PCD_DBL_BUF},
+  {CDC_OUT_EP, PMA_CDC_OUT_ADDR, PCD_SNG_BUF},
   {CDC_IN_EP,  PMA_CDC_IN_ADDR,  PCD_SNG_BUF},
   {CDC_CMD_EP, PMA_CDC_CMD_ADDR, PCD_SNG_BUF}
 #endif
```

Ulvhedin, 2021-07-27:
I can confirm above solution works: Ender 3 Pro, SKR Mini E3 v2.0, chipset GigaDevice GD32F103RET6.

MakManIfest, 2021-10-01:
I have followed this mod to the T for a Voxelab Aquila but I'm having issues with freezing and resetting when I  home Z. I am using a 3D touch setup on the probe port of the BTT SKR mini e3 v2. I've triple checked all the pins are connected correctly and figure it is something in the firmware.
Started by checking the code in the sections that were modified and discovered in pins_BTT_SKR_MINI_E3_common.h that the USB connect pin is set to PA14 which is the same as one of the signal lines on the probe port. What should I change this to?

Another issue is that the input on the screen is backward. Previously I would rotate the knob clockwise to increase number ie nozzle temp or to scroll down the menus, Now I have to rotate counter clockwise. Not a huge issue but it is annoying. How can I change that as well?

MakManIfest, 2021-10-02:
Update: Turns out the screen freezing and reset was happening after completion of homing any axis. Home X only... freeze. Home All, it homes all axis then freezes.

However, I tried applying blazewicz changes (including those found here https://github.com/MarlinFirmware/Marlin/pull/22288/files#diff-5a0102cf6df090779b130b924a2511da4922373bfc60c8c4eca427343ba6cc73R120-R124) to Alexqzd's fork V1.3.5 for the Aquila and everything seems to be working fine on the initial testing. I'll go ahead and finish applying my personal settings and run some test prints.

jrneulight, 2021-10-12:
Is the freezing issue fixed and USB issue fixed now? If so, what changes need to be made to fix those issues. I am planning on ordering an SKR Mini E3 V2 for my Ender 3 v2 to use with separate Dual Z and Octoprint. Thank you

@blazewicz

ColinM9991, 2021-11-03:
Are there any more wiring diagrams that are more comprehensible?

It is unclear what the viewing angle is of the diagram attached and the pin-out table in the common.h

```cpp
  /**
   *        ------              ------              ------
   *   VCC | 1  2 | GND    VCC | 1  2 | GND    GND |  2 1 | VCC
   *     A | 3  4 | B        A | 3  4 | B        B |  4 3 | A
   *       | 5  6   TX    BEEP | 5  6   ENT    ENT |  6 5 | BEEP
   *       | 7  8 | RX      TX | 7  8 | RX      RX |  8 7 | TX
   *  BEEP | 9 10 | ENT        | 9 10 |            | 10 9 |
   *        ------              ------              ------
   *         EXP1                DWIN             DWIN (plug)
   *
   * All pins are labeled as printed on DWIN PCB. Connect TX-TX, A-A and so on.
   */
```

Which wires exactly should be crossed?

Jaans, 2021-11-04:
> ~Are there any other wiring diagrams that are more comprehensible?~
> 
> ~It is unclear what the viewing angle is of the diagram attached and the pin-out table in the common.h~
> 
> ~Which wires exactly should be crossed? Any examples that respect the original colour coding of the Ender 3 v2 display cable?~
> 
> _Update_ Got there in the end. Soon I'll upload a colour chart for the stock display/IDC cable.

FWIW, I found the following from @Damquis to be very helpful:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/424#issuecomment-862795145
The colors used match my Creality ribbon cable colors so that made the mapping easier.

I now have a fully working Ender3v2 with the following configured:
* BTT SKR Mini E3 v2.0 motherboard 
* EZ ABL Pro 18mm probe sensor
* OctoPI working over USB
* BondTech Extruder (bowden)
* Linear Advance
* Marlin 2.0.9.2 and DWIN Enhanced (Rotary working)
_Site Note: Still working out exactly why, but the printed surface deposits are so much smoother than when I used the Creality 4.2.2 board._

A big thanks to everybody for their help and contribution!

ColinM9991, 2021-11-04:
> > ~Are there any other wiring diagrams that are more comprehensible?~
> > ~It is unclear what the viewing angle is of the diagram attached and the pin-out table in the common.h~
> > ~Which wires exactly should be crossed? Any examples that respect the original colour coding of the Ender 3 v2 display cable?~
> > _Update_ Got there in the end. Soon I'll upload a colour chart for the stock display/IDC cable.
> 
> FWIW, I found the following from @Damquis to be very helpful: [bigtreetech/BIGTREETECH-SKR-mini-E3#424 (comment)](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/424#issuecomment-862795145) The colors used match my Creality ribbon cable colors so that made the mapping easier.
> 
> I now have a fully working Ender3v2 with the following configured:
> 
> * BTT SKR Mini E3 v2.0 motherboard
> * EZ ABL Pro 18mm probe sensor
> * OctoPI working over USB
> * BondTech Extruder (bowden)
> * Linear Advance
> * Marlin 2.0.9.2 and DWIN Enhanced (Rotary working)
>   _Site Note: Still working out exactly why, but the printed surface deposits are so much smoother than when I used the Creality 4.2.2 board._
> 
> A big thanks to everybody for their help and contribution!

Thank you!

Mine was slightly different. It was 
1) Black
2) White
3) Grey
4) Purple
6) Yellow
8) Orange
9) Blue
10) Green

As it was my first time crimping an IDC connector there was a bit of confusion as to what the orientation was of the diagrams.

In the end I managed to crimp a cable and the printer is now up and running with BLTouch, just have to calibrate again and shift my calibration to the Conf file do I don't have to do that every time.

Also, a big thank you to everybody who has been looking at this!

manpres2, 2021-11-13:
Hi,

I am getting the error message while compiling 

"error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1"

ellensp, 2021-11-13:
disable EMERGENCY_PARSER in Configuration_adv.h is your only option at this time.

thisiskeithb, 2021-11-13:
> disable EMERGENCY_PARSER in Configuration_adv.h is your only option at this time.

I'd compile with the non-maple environments since `EMERGENCY_PARSER` is supported.

manpres2, 2021-11-14:
> 
> 
> disable EMERGENCY_PARSER in Configuration_adv.h is your only option at this time.

> 
> 
> > disable EMERGENCY_PARSER in Configuration_adv.h is your only option at this time.
> 
> I'd compile with the non-maple environments since `EMERGENCY_PARSER` is supported.

Hi thank you for the help and yes i disabled the emergency parser to get the firmware working.

manpres2, 2021-11-14:
i am  getting the below error messages when i try to enable BL touch bed leveling .

Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:904:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
      ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1489:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."
   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."  
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:904:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:904:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
      ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1489:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."
   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."  
    ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1489:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."
   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."  
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:904:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
      ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1489:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."
   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."  
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:904:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
      ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1489:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."
   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."  
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp:25:
Marlin\src\HAL\STM32F1\dogm\../../../inc/SanityCheck.h:904:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
      ^~~~~
Marlin\src\HAL\STM32F1\dogm\../../../inc/SanityCheck.h:1489:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."
   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."  
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:29:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:904:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
      ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1489:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, [... cut ...]
     #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a Graphical LCD."
      ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1489:4: error: #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."
   #error "Please enable only one probe option: PROBE_MANUALLY, SENSORLESS_PROBING, BLTOUCH, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."  
    ^~~~~

ellensp, 2021-11-14:
@manpres2 this is not a support forum. Please don't treat it as such,

manpres2, 2021-11-14:
> 
> 
> @manpres2 this is not a general support forum. Please don't treat it as such,

@ellensp i apologize for this but i continued the same thread as i am getting error messages in the same firmware compilation when i enable BL touch and the error messages are related to that only. Its easy for the thread members to follow it here then in a new thread. i can make a new thread if this doers not belong here.

ellensp, 2021-11-14:
Your question have nothing to do with this thread, they are simple configuration errors.
This is a feature request to add this support configuration to Marlin
Not here to help you set this up. Use the discord.

awrabb6, 2021-11-25:
> 

How can I go about local adruino framework? I can not compile my firmware with flashing from SD, I get an error saying SD flashing was only compatible with atmega2560 boards so I suspect this might fix my issue

thinkyhead, 2022-01-26:
Duplicate of #649

closed 2022-01-26 by thinkyhead