MarlinFirmware/Configurations #536: Bigtreetech profile skr 2

Giusegt, 2021-07-01:
Good evening everyone, I have created a profile for the bigtreetech skr 2 card to be mounted on Ender 3 Pro, if not my profile you could at least load one in addition to skr 1.4?

ellensp, 2021-07-25:
If you wish to submit a example config for Ender 3 Pro with a  bigtreetech skr 2 then please create a PR request for it. If you are unable than attach the files.

I cannot work out what you mean by the second part of the sentence.

thisiskeithb, 2021-07-25:
> If you wish to submit a example config for Ender 3 Pro with a bigtreetech skr 2 then please create a PR request for it. If you are unable to create a PR then attach the files.
> 
> I cannot work out what you mean by the second part of the sentence.

"ghost" accounts are from users who deleted their account, so this feature request can be closed.

closed 2021-07-25 by thisiskeithb