MarlinFirmware/Configurations #542: Creality Sermoon D1 - BLTouch Support

Valexxia, 2021-07-16:
### Description

Uses the same board as the ender 6 however it seems that creality simply has no support for the sermoon D1, It does have a extension board with a bltouch port on it however there is no firmware from the standard 1.1.1 that comes preloaded on the machine.

### Configuration Details

apoligies for the lack of information, i am happy to try and find anything that is needed. just not sure what to look for

* Manufacturer: Creality
* Model: Sermoon D1
* Motherboard: Creality 4.3.1

#### Additional Information

* Provide links to the requested printer/configuration.

thinkyhead, 2021-07-16:
You may be able to enable pins-debugging and then use M43 to figure out which pins are associated with the BLTouch.

Valexxia, 2021-07-16:
oh ok, how would i go about enabling pin debugging ?

tpruvot, 2021-07-16:
Marlin/Configuration_adv.h
```
// M43 - display pin status, toggle pins, watch pins, watch endstops & toggle LED, test servo probe
//
#define PINS_DEBUGGING

```

Valexxia, 2021-07-16:
sorry to be a pain, the firmware file is a .bin file, how would i go about opening that to be able to edit it ?

ellensp, 2021-07-17:
you don't edit bin files.
You edit Configuration files and recompile

Valexxia, 2021-07-17:
ok great to know, but how do i get the configuration files to begin with ?

ellensp, 2021-07-17:
You  ask Creality nicely for them, and when they refuse, you remind them they are in violation of the licensing and ask again.
When they refuse that  you set the lawyers on them.

ellensp, 2021-07-17:
Then you open up the machine and trace all the wires and document what is plugged into where.

Valexxia, 2021-07-17:
so if thats the case would tham mean that i could use the configuration files from the ender 6 as its the same board used between both machines or would that complicate things too much ?

ellensp, 2021-07-17:
The Config files describes way more than just the motherboard. Egs build size, direction to move the steppers, homing direction. LCD.Screen settings 
Au though it will boot up if it is the same motherboard and talk over serial/usb it is unlikely to do anything positove for your machine and could damage things

Valexxia, 2021-07-17:
ok, well i messaged Creality two days ago so if i hear back from them ill post it up here

mcecchi, 2021-08-11:
Hi @Valexxia, any updates? I wrote to Creality today for my CR-200B to claim Marlin source code.

Valexxia, 2021-08-11:
No updates yet, I have sent them multiple messages and emails. They just don’t reply

Randy1727, 2021-10-17:
@ellensp Have you managed to compile BL touch enabled in firmware using breakout board on Sermoon? If so do you mind sharing?

ellensp, 2021-10-17:
@Randy1727 with the patches above if you enbale #define BLTOUCH, it still compiles
I dont have any of this hardware, I cant test it beyond compiling

InsanityAutomation, 2021-10-17:
I've got a current code base synced as of Friday that has the board and touchscreen working, but I don't have a Sermoon to physically test on. I'll be cleaning things up over the next day or two and can drop it over if you want to test.

Randy1727, 2021-10-19:
@InsanityAutomation  Being im a total noob when it comes to changing confg files and compiling firmware please excuse my ignorance.  The only things I enabled are #define MachineSermoonD1   #define HotendStock    #define DirectDrive    #define BedDC    #define ABL_BLTOUCH  #define GraphicLCD   #define MeshFine, Is there anything else that needs to activated for the Sermoon, Also In the zip there are 3 DWIN folders, I take it the Sermoon will use the Combinedscreens one?

Thank you for your help, Im still learning/.

Randy1727, 2021-10-20:
@InsanityAutomation  These are the errors im getting 
------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Environment: LPC1769
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.9
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
Building in release mode
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_MinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\xpt2046.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\watchdog.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\HAL_MinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\servo.cpp.o
In file included from Marlin\src\HAL\LPC1768\../../inc/../pins/stm32f1/pins_CREALITY_V4.h:28,
                 from Marlin\src\HAL\LPC1768\../../inc/../pins/pins.h:547,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:34,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/../pins/stm32f1/env_validate.h:25:4: error: #error "Oops! Select an STM32F1 board in 'Tools > Board.'"
   25 |   #error "Oops! Select an STM32F1 board in 'Tools > Board.'"
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/../pins/stm32f1/pins_CREALITY_V4.h:28,
                 from Marlin\src\HAL\LPC1768\../../inc/../pins/pins.h:547,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:34,
                 from Marlin\src\HAL\LPC1768\HAL_MinSerial.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/../pins/stm32f1/env_validate.h:25:4: error: #error "Oops! Select an STM32F1 board in 'Tools > Board.'"
   25 |   #error "Oops! Select an STM32F1 board in 'Tools > Board.'"
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:681:6: error: #error "Use SERIAL_PORT_2 before using SERIAL_PORT_3"
  681 |     #error "Use SERIAL_PORT_2 before using SERIAL_PORT_3"
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/../pins/stm32f1/pins_CREALITY_V4.h:28,
                 from Marlin\src\HAL\LPC1768\../../inc/../pins/pins.h:547,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:34,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\LPC1768\../../inc/../pins/stm32f1/env_validate.h:25:4: error: #error "Oops! Select an STM32F1 board in 'Tools > Board.'"
   25 |   #error "Oops! Select an STM32F1 board in 'Tools > Board.'"
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\HAL_MinSerial.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:681:6: error: #error "Use SERIAL_PORT_2 before using SERIAL_PORT_3"
  681 |     #error "Use SERIAL_PORT_2 before using SERIAL_PORT_3"
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/../pins/stm32f1/pins_CREALITY_V4.h:28,
                 from Marlin\src\HAL\LPC1768\../../inc/../pins/pins.h:547,
[... 418 lines cut ...]
      |    ^~~~~
Marlin\src\inc/../HAL/LPC1768/inc/SanityCheck.h:168:6: error: #error "Serial port pins (3) conflict with X endstop pins!"
  168 |     #error "Serial port pins (3) conflict with X endstop pins!"
      |      ^~~~~
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\tft\xpt2046.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o] Error 1
Marlin\src\MarlinCore.cpp:1172:8: error: #error "DISABLE_(DEBUG|JTAG) is not supported for the selected MCU/Board."
 1172 |       #error "DISABLE_(DEBUG|JTAG) is not supported for the selected MCU/Board."
      |        ^~~~~
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\watchdog.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\shared\Delay.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\shared\HAL_MinSerial.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\shared\backtrace\backtrace.cpp.o] Error 1
[... cut ...]
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1769\src\src\MarlinCore.cpp.o] Error 1
=================================================================== [FAILED] Took 2.97 seconds ===================================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:02.968
============================================================= 1 failed, 0 succeeded in 00:00:02.968 =============================================================

Randy1727, 2021-10-20:
Got it working, compiled and screen is working. Im a idiot and didnt see the zip of precompiled firmware.

Randy1727, 2021-10-20:
Ok, Installed the firmware and dwin from combined screens v5.7 and some buttons such as move, motor control dont work, Level and info do come up but unable to back out.

Randy1727, 2021-10-20:
@InsanityAutomation Language and printer info work and let you back out, Move, motor, refuel, print and temp dont work at all, Leveling lets you enter but not back out. Movement commands from octoprint do work, I sent you a msg on discord,

ernanilm, 2021-10-25:
I am waiting for this firmware

mikeroxy2019, 2021-11-04:
heyo, im also kind of a noob, but my D1 sermoon is coming tomorrow in the mail.  Ive been reading up, glad to see creality released the source.  I grabbed it and changed to the older platformio configs above and got it to compile. But I see the lcd isn't loving you guys.  Can I ask how the progression goes ?  I downloaded the latest hex files from insanity but im reluctant to flash tomorrow if my screen won't work right and id be stuck trying to use octprint solely. thanks again for your awesome work !

InsanityAutomation, 2021-11-04:
Still a few more things to work out, so I'd hold off for a little while. Unfortunately day job has me working all weekend again so it'll be a little bit until I can wrap it up.

mikeroxy2019, 2021-11-04:
No rush man.  I’m excited that you’ll get it working soon. Glad I got the printer once the source code was actually released.  Crazy that these high end creality printers don’t come with an ABL sensor in the box. Or official downloadable factory firmware to add one.   Creality should pay you!

meddle99, 2021-11-04:
I designed a mount for the BL touch on the sermoon.  The regular mounts didn't work.  https://www.thingiverse.com/thing:5050263
I've had my sermoon for 2 weeks and now I think the main board is dead.  It worked perfectly up until the weekend, but now any print longer than an hour seems to crash.

mikeroxy2019, 2021-11-04:
I already downloaded the mount a few hours ago ;)  I’m reading that this 4.3.1 board isn’t so great.  Only has silent steppers on x and y?  I’ll refrain from what I think about that.  Think it’s worth it to swap it for a 4.2.7?  Would there be any problems with firmware .. besides swapping board declaration?

meddle99, 2021-11-04:
There are many people on this board who are imminently more qualified to answer that question, I'm just a newbie in this area. Although if I were to buy a new board I'm not sure I would get the creality.

InsanityAutomation, 2021-11-04:
I did find the power supply was drastically undersized on the Ender 7 causing occasional shutdowns when the driver pulled the voltage down too low... First I would check cooling on the main board and then the power supply before assuming the board is dead.

meddle99, 2021-11-04:
Any idea what the power capacity was on the undersized supply?  I just checked and the one in the D1 is 24V 14.6A.  I'll hook up my meters and measure input and output current during a print tomorrow.

meddle99, 2021-11-05:
While the bed and hot end were heating, I measured 12.5A.  While printing, current stayed around 6A most of the time, but peaked a couple of times to 13.  Maybe it's not in the danger zone, but a little undersized for my comfort.  I don't like seeing sustained current over 80% of the capacity.  

@mikeroxy2019, even without the silent stepper on the X, the printer is still very quiet.  I don't think it's a big deal.

ETE-Design, 2021-12-16:
@thinkyhead Will the Sermoon D1 be added to Configuration soon? Or is there still a lot of work to do on the Screen Part?

InsanityAutomation, 2021-12-17:
@ETE-Design its up here https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0

rmurilo, 2021-12-24:
> Still a few more things to work out, so I'd hold off for a little while. Unfortunately day job has me working all weekend again so it'll be a little bit until I can wrap it up.

Any luck on getting this working I would love to add the touch to it and have u any idea how to do the e steps o can we really not do it on this machine

InsanityAutomation, 2021-12-25:
@rmurilo see the link in the comment above, code is up and working. I'll be adding the cr5 and cr10 smart shortly as well hopefully.

rmurilo, 2021-12-25:
Sorry so just to confirm the same file works for all the machines listed I don’t see the sermoon d1 listed in there from the link above could u post a direct link to the specific download for the sermoon d1 please and happy holidays

rmurilo, 2021-12-26:
I joined the discord group thank you for all you do for these machines

ETE-Design, 2022-01-02:
@InsanityAutomation Is it possible to add BLTouch with your firmware? And if so, where to find a wire diagram for it?

mikeroxy2019, 2022-01-02:
There’s a pre compiled bin file with bltouch support on IA’s GitHub already.  Board uses the same pins as a 4.2.7 / 4.2.2.

ETE-Design, 2022-01-29:
@InsanityAutomation The Screen on the Sermoon D1 is it a DWIN T5UID1?

FeLiXXX81, 2022-02-18:
can someone please help me with compiling firmware for sermoon d1?
I downloaded it from creality but it have errors when I try to compile surce code.
[image]

> Executing task: C:\Users\FeLiXXX\.platformio\penv\Scripts\platformio.exe run <
`> Executing task: C:\Users\FeLiXXX\.platformio\penv\Scripts\platformio.exe run <

Processing creality (board: genericSTM32F103RE; platform: ststm32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (15.2.0) > STM32F103RE (64k RAM. 512k Flash)       
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.20100.211028 (2.1.0)
 - framework-cmsis 2.50700.210515 (5.7.0)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 20 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.3
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.7.3
|   |-- <SoftwareSerial> 1.0
|-- <Adafruit MAX31865 library> 1.3.0
|   |-- <Adafruit BusIO> 1.9.1
|   |   |-- <Wire> 1.0
|-- <LiquidTWI2> 1.2.7
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD> 0.0.0+20210827085153
|-- <SailfishRGB_LED> 0.0.0+20210827085153
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster> 0.0.0+20210827085154
|-- <EEPROM> 2.0.1
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\creality\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\creality\FrameworkArduinoVariant\PeripheralPins_STORM32_V1_31_RC.c.o
Compiling .pio\build\creality\FrameworkArduinoVariant\generic_clock.c.o
Compiling .pio\build\creality\FrameworkArduinoVariant\variant_BLUEBUTTON_F103RxT.cpp.o
Compiling .pio\build\creality\FrameworkArduinoVariant\variant_STORM32_V1_31_RC.cpp.o
Compiling .pio\build\creality\FrameworkArduinoVariant\variant_generic.cpp.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_adc.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_can.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_cec.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_comp.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_comp_ex.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_cordic.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_cortex.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_crc.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_crc_ex.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_cryp.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_cryp_ex.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_dac.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_dac_ex.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_dcache.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_dcmi.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_dcmi_ex.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_dfsdm.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_dfsdm_ex.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_dma.c.o
Compiling .pio\build\creality\SrcWrapper\src\HAL\stm32yyxx_hal_dma2d.c.o
[... 303 lines cut ...]
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\eeprom_i2c.cpp:28:
Marlin\src\HAL\shared\../../inc/../HAL/./HAL_STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\creality\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o] Error 1
*** [.pio\build\creality\src\src\HAL\shared\eeprom_i2c.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./HAL_STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\eeprom_spi.cpp:28:
Marlin\src\HAL\shared\../../inc/../HAL/./HAL_STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\creality\src\src\HAL\shared\eeprom_spi.cpp.o] Error 1
=============================================================================================== [FAILED] Took 8.13 seconds ===============================================================================================
Environment    Status    Duration
-------------  --------  ------------
creality       FAILED    00:00:08.128
========================================================================================= 1 failed, 0 succeeded in 00:00:08.128 ========================================================================================= 
The terminal process "C:\Users\FeLiXXX\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

thinkyhead, 2022-03-20:
You should ignore messages under the "Problems" tab, which are generated by other VSCode extensions.

thinkyhead, 2022-04-10:
Duplicate of #649

closed 2022-04-10 by thinkyhead