MarlinFirmware/Configurations #544: BLTouch A10M config not working

mrtron28, 2021-07-21:


ellensp, 2021-07-25:
There is no BLTOUCH A10M config in https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Geeetech/A10M 

So how can it be not working?

closed 2021-07-25 by ellensp

mrtron28, 2021-07-25:
i edit the config and use BlTouch on. BlTouch lights and  geeetech a10 3d
printer runs in middle of bed. Bltouch light red. self test working....but
no homeing down towards bed....i think a bug but not sure....

thisiskeithb, 2021-07-25:
> i think a bug but not sure....

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).