MarlinFirmware/Configurations #549: Ender 3 V2 with Creality 4.2.2 board E0 does not work when enable Linear Advance

jeremy32704, 2021-08-01:
### Description
I am new to 3D printing and have tried multiple solutions and tips I have found to resolve this issue, but so far no luck.  I have a stock Ender 3 V2 with no upgrades, YET, but when trying to enable Linear Advance my E0 stepper motor will start moving on the initial clean nozzle move but when it moves over to come back it stops working and does not move for the rest of the Linear Advance test print or even if I try to print a test block.  I have tried the latest bugfix 2.0.x configuration and Marlin files, I have tried 2.0.8 and 2.0.9 version files.  With all of these the E0 stepper will work if I disable Linear Advance just like with Crealitys firmware, but when I try to enable Linear Advance which is why I am trying to update firmware anyway it does not work.  This could be a known setup issue that I have not been able to find yet, although I have searched google and github trying to find a solution for this and tried all tips with no success yet.  If anyone could help with my configuration to get this to work, that would be GREAT.

### Configuration Details

* Manufacturer: Creality 
* Model: Ender 3 V2
* Motherboard: Creality STM32F103RET6 4.2.2

#### Additional Information
The stock stepper motor driver on this board is the TMC 2208
* Provide links to the requested printer/configuration.

jeremy32704, 2021-08-01:
I forgot to mention that I was able to move the E0 stepper motor from the move axis menu selection and verify calibration with each firmware version I tried out.

jeremy32704, 2021-08-01:
I uploaded the configuration.h and _adv.h files here for review

[Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/6913019/Marlin.zip)

thisiskeithb, 2021-08-01:
This is a known issue with Linear Advance and TMC2208s in standalone mode on Creality boards (just search his issue queue).

Disable LA or replace the motherboard with one that doesn’t use 2208s in standalone mode.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-08-01 by thisiskeithb

jeremy32704, 2021-08-01:
Thanks for your reply.  Is there any known issues with Linear advance and TMC 2225 drivers?  I will look to purchase the creality 4.2.7 board.

thisiskeithb, 2021-08-01:
> Thanks for your reply. Is there any known issues with Linear advance and TMC 2225 drivers? I will look to purchase the creality 4.2.7 board.

TMC2225s are just TMC2208s in a different package. You'd want TMC2209s or TMC2226s.