MarlinFirmware/Configurations #554: ender 3 with creality touchscreen (4.2.7 board)

eagle00789, 2021-08-15:
there should be a working config for the ender 3 with 4.2.7 board with support for the newly released touchscreen (based on the CR-6 also without rotary encoder)
The touchscreen is mentioned [here](https://www.creality3dofficial.com/collections/accessories/products/upgraded-touch-panel-screen-display-for-ender-3-pro-ender-3-v2) and this page mentions the new touchscreen with 4.2.7 support based on DWIN

thisiskeithb, 2021-08-16:
That TFT is not supported in Marlin currently, but will once https://github.com/MarlinFirmware/Marlin/pull/19958 is merged.

thinkyhead, 2021-10-30:
That PR has been stalled pending testing. I have a CR-6 SE I can play around with here, so I will make a special effort to get that merged soon.

thinkyhead, 2022-04-10:
Duplicate of #649

closed 2022-04-10 by thinkyhead

dhysuiej, 2023-10-21:
@eagle00789 I'm also interested in this. Mine has the 4.2.2 board. Did you manage to get the CR-6 touch screen working on Ender 3 running Marlin.