MarlinFirmware/Configurations #557: [FR] Bump configurations to 2.0.9.1 - currently at 2.0.8.2

bryanpedini, 2021-08-19:
### Description

Is it possible to bump the configurations to the latest [MarlinFirmware/Marlin](https://github.com/MarlinFirmware/Marlin) release? Or does 2.0.8.2 work just fine?
Thanks!

cp2004, 2021-08-19:
They are tagged, they just have no release notes:
[image]

You can also find all the config versions by branch here:
[image]

thisiskeithb, 2021-08-19:
See https://github.com/MarlinFirmware/Configurations/releases/tag/2.0.9.1

closed 2021-08-19 by thisiskeithb

bryanpedini, 2021-08-19:
Oopsie, sorry.
Idiot me not looking thorougly.

bchanudet, 2021-09-09:
Hello there,

May I open again this issue? Because the latest releases are not tagged as the others, they also don't appear on the REST API:

Getting the list of releases doesn't return at all the `2.0.9` releases: https://api.github.com/repos/MarlinFirmware/Configurations/releases

bryanpedini, 2021-09-09:
does it not appearing on the APIs matter?
doesn't GH also provide APIs for tags? 😅

bryanpedini, 2021-09-09:
> https://api.github.com/repos/MarlinFirmware/Configurations/tags

Damn it, you stole my scene... I was reading the GH documentation about my last question 😝

bryanpedini, 2021-09-09:
@bchanudet TLDR: use the link provided above instead, it also returns `{tar,zip}balls` links for download like the releases (of course, not being releases themselves, only source code downloads are available, anything uploaded by the maintainers on that release is not, because there's no release about that)

bchanudet, 2021-09-09:
@bryanpedini: completely forgot about tags, thanks for the reminder!

I'm currently making a script to keep my 3D Printer up-to-date automatically with the latest Marlin version, and by default I used the `/releases/latest` endpoint for the firmware repo, and I was puzzled why I couldn't fetch the corresponding release on `/releases/tags/2.0.9.1` for the Configurations repo.

I will adjust my code to use the tags endpoints instead. 

Thanks for you quick reply!

bryanpedini, 2021-09-09:
> @bryanpedini: completely forgot about tags, thanks for the reminder!

No prob, actually it was The-EG who reminded you, but whatever 😅 

> I'm currently making a script to keep my 3D Printer up-to-date automatically with the latest Marlin version

NOOOOO, what have you dooooone?
Seriously: good luck with fresh new released bugs then :)

bchanudet, 2021-09-09:
Let me thanks @The-EG then too! :)

> Seriously: good luck with fresh new released bugs then :)

"Automatically" was more like "Script that I can run next year when I want to update the firmware so I don't have to remember how to do it manually" than "Run it and upgrade in a CI/CD manner". I do want to use my printer sometimes :p

bryanpedini, 2021-09-09:
> I do want to use my printer sometimes :p

Haaaaah, now thaaat is better sounding 😅