MarlinFirmware/Configurations #560: NeoPixel

LunerStorm, 2021-08-30:
* Manufacturer: Creality
* Model: Ender 5
* Motherboard: SKR mini e3 v2
[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/7080032/Configuration.zip)

#### Additional Information

* Provide links to the requested printer/configuration.

thisiskeithb, 2021-08-30:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-08-30 by thisiskeithb

LunerStorm, 2021-08-31:
Hi Keith  Pretty sure this is a bug as I can get it working in 2.0.6. However 2.0.9.1 compiles but does not work.DonSent from my Verizon, Samsung Galaxy smartphone

thisiskeithb, 2021-08-31:
> Hi Keith  Pretty sure this is a bug as I can get it working in 2.0.6. However 2.0.9.1 compiles but does not work.DonSent from my Verizon, Samsung Galaxy smartphone

Please file a proper bug report in the correct repo: [MarlinFirmware/Marlin](https://github.com/MarlinFirmware/Marlin/issues/new?assignees=&labels=&template=bug_report.yml&title=%5BBUG%5D+%28bug+summary%29).