MarlinFirmware/Configurations #561: [FR] XYZPrinting Da Vinci

kubark42, 2021-09-01:
### Description

A number of people have an XYZPrinting Da Vinci printer, and while it has a Repetier fork bouncing around, it's quite old now and the Marlin firmware would be a very nice upgrade.

The Da Vinci is a nicely build enclosed printer which uses a 32-bit Atmel ATSAM3X8E microcontroller and A4988ET stepper motor drivers.

### Configuration Details

* Manufacturer: XYZPrinting
* Model: Da Vinci 1.0
* Motherboard: 1.0
* Repetier fork: https://github.com/luc-github/Repetier-Firmware-4-Davinci

#### Additional Information

The above linked Repetier fork presumably has all the pin configurations, if someone is familiar with how to read Repetier configurations.

kubark42, 2021-09-01:
I'd be very happy to take point on this if someone can walk me through the process of bringing up a new board. I did this a lot at Tau Labs, so am familiar with what it takes in general. I lack the specifics of how to do this with Marlin.

ellensp, 2021-09-01:
Its not just a board...
The LCD is also non standard. a WINSTAR 16x4 char display.  and 5 buttons, but how is it all connected?

kubark42, 2021-09-01:
I feel like the LCD is a nice-to-have, but certainly not a requirement. If there's already support in Marlin for this panel and it's only the pins which change, then cool. Otherwise, arguably it's not worth adding code bloat for a non-standard screen.

Great question about how it's all connected. I did some digging in the Repetier fork and found several interesting documents. I linked them to my top post. It has uC pinouts, LCD documentation, etc...

The LCD is connected with a thin ribbon cable, as show in the pic below:

[image]

ellensp, 2021-09-04:
Well the first thing in my view is to examine the script apply_patches.py, work out what it does

It updates .platformio/packages/framework-arduino-sam with specifics for this board,

It replaces 
.platformio/packages/framework-arduino-sam/cores/arduino/USB/USBCore.cpp
with the file Repetier-Firmware-4-Davinci/PIOPatches/USBCore.cpp
and 
.platformio/packages/framework-arduino-sam/variants/arduino_due_x/variant.cpp 
with the file Repetier-Firmware-4-Davinci/PIOPatches/variant.cpp

```DIFF
diff Repetier-Firmware-4-Davinci/PIOPatches/USBCore.cpp .platformio/packages/framework-arduino-sam/cores/arduino/USB/USBCore.cpp
202,208c202
< //USB PATCH to avoid dead loop
< 		int count=0;
<     while( UOTGHS_DEVEPTISR_TXINI != (UOTGHS->UOTGHS_DEVEPTISR[ep & 0xF] & UOTGHS_DEVEPTISR_TXINI ))
<         {
<             count++;
<             if (count>10000) return len;
<         }
---
>
```

```DIFF
diff Repetier-Firmware-4-Davinci/PIOPatches/variant.cpp .platformio/packages/framework-arduino-sam/variants/arduino_due_x/variant.cpp
20c20
< //For Davinci Printer
---
> 
271,273c271,273
<     // 85 - USB
<   //{ PIOB, PIO_PB11A_UOTGID|PIO_PB10A_UOTGVBOF, ID_PIOB, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL,NO_ADC, NO_ADC, NOT_ON_PWM,  NOT_ON_TIMER }, // ID - VBOF
<   { PIOB, PIO_PB11A_UOTGID, ID_PIOB, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL,NO_ADC, NO_ADC, NOT_ON_PWM,  NOT_ON_TIMER }, // ID - VBOF
---
>   // 85 - USB
>   { PIOB, PIO_PB11A_UOTGID|PIO_PB10A_UOTGVBOF, ID_PIOB, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL,NO_ADC, NO_ADC, NOT_ON_PWM,  NOT_ON_TIMER }, // ID - VBOF
> 
290,334c290
<   // 92 .. 99 placeholders, future-proofing.
<  // { NULL, 0, 0, PIO_NOT_A_PIN, PIO_DEFAULT, 0, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER },
<  // { NULL, 0, 0, PIO_NOT_A_PIN, PIO_DEFAULT, 0, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER },
<   { PIOB, PIO_PB22,          ID_PIOB, PIO_OUTPUT_0, PIO_DEFAULT, PIN_ATTR_DIGITAL,                  NO_ADC, NO_ADC, NOT_ON_PWM,  NOT_ON_TIMER }, // PIN 92 - PB22 E1 Enabled
<   { PIOB, PIO_PB10, ID_PIOB, PIO_PERIPH_B, PIO_DEFAULT, PIN_ATTR_DIGITAL,NO_ADC, NO_ADC, NOT_ON_PWM,  NOT_ON_TIMER }, // ID  9
<   { NULL, 0, 0, PIO_NOT_A_PIN, PIO_DEFAULT, 0, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER },
<   { NULL, 0, 0, PIO_NOT_A_PIN, PIO_DEFAULT, 0, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER },
<   { NULL, 0, 0, PIO_NOT_A_PIN, PIO_DEFAULT, 0, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER },
<   { NULL, 0, 0, PIO_NOT_A_PIN, PIO_DEFAULT, 0, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER },
<   { NULL, 0, 0, PIO_NOT_A_PIN, PIO_DEFAULT, 0, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER },
<   { NULL, 0, 0, PIO_NOT_A_PIN, PIO_DEFAULT, 0, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER },
<   // 100 - 108 extra SAM3X8E pins, not wired on Due
<   { PIOA, PIO_PA5, ID_PIOA, PIO_OUTPUT_0, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN 100
<   { PIOC, PIO_PC27, ID_PIOC, PIO_OUTPUT_0, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN 101
<   { PIOA, PIO_PA0, ID_PIOA, PIO_OUTPUT_0, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN 102
<   { PIOA, PIO_PA1, ID_PIOA, PIO_OUTPUT_0, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN 103
<   { PIOC, PIO_PC11, ID_PIOC, PIO_OUTPUT_0, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN 104
<   { PIOC, PIO_PC8B_PWML3, ID_PIOC, PIO_PERIPH_B, PIO_DEFAULT, (PIN_ATTR_DIGITAL|PIN_ATTR_PWM), NO_ADC, NO_ADC, PWM_CH3, NOT_ON_TIMER }, // PWM 105
<   { PIOC, PIO_PC2B_PWML0, ID_PIOC, PIO_PERIPH_B, PIO_DEFAULT, (PIN_ATTR_DIGITAL|PIN_ATTR_PWM), NO_ADC, NO_ADC, PWM_CH0, NOT_ON_TIMER }, // PWM 106
<   { PIOC, PIO_PC6B_PWML2, ID_PIOC, PIO_PERIPH_B, PIO_DEFAULT, (PIN_ATTR_DIGITAL|PIN_ATTR_PWM), NO_ADC, NO_ADC, PWM_CH2, NOT_ON_TIMER }, //PWM 107
<   { PIOC, PIO_PC20, ID_PIOC, PIO_OUTPUT_0, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, //PWM 108
<   // 109 .. 114
<   { PIOA, PIO_PA20A_MCCDA, ID_PIOA, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN_HSMCI_MCCDA_GPIO 109
<   { PIOA, PIO_PA19A_MCCK, ID_PIOA, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN_HSMCI_MCCK_GPIO 110
<   { PIOA, PIO_PA21A_MCDA0, ID_PIOA, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN_HSMCI_MCDA0_GPIO 111
<   { PIOA, PIO_PA22A_MCDA1, ID_PIOA, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN_HSMCI_MCDA1_GPIO 112
<   { PIOA, PIO_PA23A_MCDA2, ID_PIOA, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN_HSMCI_MCDA2_GPIO 113
<   { PIOA, PIO_PA24A_MCDA3, ID_PIOA, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // PIN_HSMCI_MCDA3_GPIO 114
<   // 115 .. 124 - ETHERNET MAC
<   { PIOB, PIO_PB0A_ETXCK, ID_PIOB, PIO_PERIPH_A, PIO_DEFAULT, PIN_ATTR_DIGITAL, NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER }, // ETXCK 115
[... 29 lines cut ...]
PIO_DEFAULT, 
PIN_ATTR_DIGITAL,
NO_ADC, 
NO_ADC, 
NOT_ON_PWM,  
NOT_ON_TIMER
 }, // ID - VBOF

Original.
>   {
PIOB, 
PIO_PB11A_UOTGID|PIO_PB10A_UOTGVBOF, 
ID_PIOB, 
PIO_PERIPH_A, 
PIO_DEFAULT, 
PIN_ATTR_DIGITAL,
NO_ADC, 
NO_ADC, 
NOT_ON_PWM,  
NOT_ON_TIMER
 }, // ID - VBOF

And then there is the "USB PATCH to avoid dead loop" to consider.

I suspect will need to create out own marlin board variant to support this board.

kubark42, 2021-09-05:
So would it be fair to say this is not as straightforward as appropriately configuring  `Configuration.h` and `Configuration_adv.h`?

If so, that's disappointing, and arguably not worthwhile until there's a larger pool of people who'd benefit.

ellensp, 2021-09-05:
Needs a board variant created, need a new board pins.h file added, needs a new type of LCD type added.
Then you can add a Configuration.h and Configuration_adv.h that uses these features

kubark42, 2021-09-06:
> Needs a board variant created, need a new board pins.h file added, needs a new type of LCD added.
> Then you can add a Configuration.h and Configuration_adv.h that uses these features

So, here's my understanding:

* New variant
  * Where are the variants kept? I have perused https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src and nothing jumps out at me.
* New `pins.h`
  * This seems easy and straightforward: Clone a ATSAM3X8E board, e.g. https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/pins/sam/pins_ARCHIM1.h, and point toward the Da Vinci board pins.
* LCD
  * Ignore LCD for the time being, as it's not a priority and is not required
* Add `Configuration.h` and `Configuration_adv.h`. 
  * This seems like it requires some amount of specialized familiarity. If I get things as far as this, is it easy to find someone to guide me at that point?

ellensp, 2021-09-08:
I have created a Marlin that complies for this board.
It creates the Variant
It has a boards pins.h file
LCD is not supported at all
It has a rudimentary Configuration files 

### I have no way to test any of this.

### Test each function with extreme caution!

It may not boot, it may set all heaters to max and try and burn you, all the axes are probably backward.

You probably should not do this is you cannot restore the machine to its current state.

NB the thermistor type are wrong...  at default of 1.   

Its a start
https://github.com/ellensp/Marlin/tree/Da_Vinci_1.0

ellensp, 2021-09-08:
You can see whats been changed in https://github.com/ellensp/Marlin/commit/9b5c44aa68427ea807c8cd57774dfbf8a3883942

kubark42, 2021-09-08:
Thanks! I've got the board pulled out of the machine and sitting on my desk, so it should be easy to test.

So that I know how to start massaging these values myself, how did you determine what changes to make? I notice that some things, such as https://github.com/ellensp/Marlin/commit/9b5c44aa68427ea807c8cd57774dfbf8a3883942#diff-bb298d18dac4268225a7a53a5ad24844472530bdc63f6a98a2ba8d8158b75080R1375, seem very precise.

ellensp, 2021-09-08:
Information was extracted from https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Configuration.h where possible

kubark42, 2021-09-08:
Ahhhh, makes sense. I'm guessing you've done this before so you knew how to translate from Repetier settings to Marlin. 🗡️ 
Okay, gonna get right on this today and see how far I can get. Hopefully because there's already Repetier on it I can easily use the bootloader to switch to Marlin.

ellensp, 2021-09-08:
Its sort of readable, but no I've have very little to do with Repetier. I just read through it, picking out what I could see translated easily to marlin

kubark42, 2021-09-08:
Gotcha.

BTW, hats off to whoever created [Auto Build Marlin](https://marlinfw.org/docs/basics/auto_build_marlin.html). I just discovered it and it was everything I could have wanted. The only, admittedly very minor, challenge was realizing that I needed to check-out the `Da_Vinci_1.0` branch in order to have ABM see the correct build files.

kubark42, 2021-09-08:
***GRUMBLE, GRUMBLE***, first hiccup is that the board won't power on with a pure USB connection, so I have to bring it back to my lab, where the rest of the printer is, before I can test any further.

kubark42, 2021-09-11:
Well, it didn't exactly brick the board since the bootloader still works, but once past the bootloader stage the USB port disappears. I haven't started to look into it yet, but I suspect that Marlin firmware uses a set of alternative serial port pins from the ones used on the board.

kubark42, 2021-09-11:
@ellensp, do you think the serial port issue could be because https://github.com/ellensp/Marlin/commit/9b5c44aa68427ea807c8cd57774dfbf8a3883942#diff-bb298d18dac4268225a7a53a5ad24844472530bdc63f6a98a2ba8d8158b75080R105 sets the `SERIAL_PORT` to `-1`?

ellensp, 2021-09-11:
when the usb port is directly connected to the Controller cpu, SERIAL_PORT should be  -1
If its using a uart with a usb to serial chip, then its ports 0-3, but I don't see any usb-serial chips on your board...not on the picture above.

Trace the D-,D+ pins from usb port, where do they go? 

On the due its these pins
D- pin 38 DHSDM
D+ pin 37 DHSDP

kubark42, 2021-09-11:
https://github.com/ellensp/Marlin/commit/9b5c44aa68427ea807c8cd57774dfbf8a3883942#diff-1cc05ed1b31b129e4b57e0d9f1da1ad3bf9aa261f05c5f09c84f1b4473ffdd8dR124 shows that the serial port is configured as PA8/PA9, which is the same as in the pinout.xls file. 

I'll trace the pins and report back to be sure.

kubark42, 2021-09-11:
Scoped out and confirmed that D- goes to 38 and D+ to pin 37.

kubark42, 2021-09-11:
Scoped out and confirmed that D- goes to 38 and D+ to pin 37.

Could this be a crystal configuration issue? The Davinci board uses a 12.000H40 crystal, which is the same 12MHz xtal as on the Archim I have here.

https://github.com/ellensp/Marlin/blob/9b5c44aa68427ea807c8cd57774dfbf8a3883942/buildroot/share/PlatformIO/variants/MARLIN_ARCHIM/variant.h shows 12MHz, but I can't find the equivalent for this board setting.

ellensp, 2021-09-12:
Due also has 12mhz crystal on xin,xout

kubark42, 2021-09-12:
Hmmm..., I am too unfamiliar with the ATSAM to have many other ideas why the chip might have crashed or the serial port might be misconfigured.

I'm thinking it might make sense to shotgun this with the variants of DUE boards and see if anything boots up properly. With a bit of luck, it would give us a known good scaffold upon which to progress.

Thoughts?