MarlinFirmware/Configurations #564: TELA DA SAPPHIRE PLUS INVERTIDA (CABEÇA PARA BAIXO)

ngoliveira, 2021-09-06:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ngoliveira, 2021-09-06:
> 

> [``](url)

willngton, 2021-09-07:
You haven't upload your config files, please check if the fallowing directive are enable:

#define TFT_ROTATION TFT_ROTATE_180

If now, just add and it should be fine.

thisiskeithb, 2021-09-07:
Do you have a V1 or V2 [Two Trees Sapphire Plus](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Two%20Trees/Sapphire%20Plus)?

We don't have configs for the V1.

`TFT_ROTATION` is disabled for for the V2 config (so not rotated), but as @willngton noted, enable `#define TFT_ROTATION` and set it to `TFT_ROTATE_180` since your TFT appears to be 180 degrees out.

We'll need someone with a V1 and V2 to confirm which settings are correct and update settings/readmes accordingly.

ngoliveira, 2021-09-08:
> Você tem um V1 ou V2 [Two Trees Sapphire Plus](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Two%20Trees/Sapphire%20Plus) ?
> 
> Não temos configurações para o V1.
> 
> `TFT_ROTATION`está desabilitado para a configuração V2 (portanto, não girado), mas como @willngton observou, habilite `#define TFT_ROTATION`e defina-o como `TFT_ROTATE_180`já que seu TFT parece estar 180 graus fora.
> 
> Precisaremos de alguém com V1 e V2 para confirmar quais configurações estão corretas e atualizar as configurações / leia-me de acordo.

A minnha Sapphire é a V2... Dois motores no Z...
Vou tentar fazer a alteração e aviso se foi resolvido...

Obrigado pela ajuda!

ngoliveira, 2021-09-08:
> Você tem um V1 ou V2 [Two Trees Sapphire Plus](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Two%20Trees/Sapphire%20Plus) ?
> 
> Não temos configurações para o V1.
> 
> `TFT_ROTATION`está desabilitado para uma configuração V2 (portanto, não girado), mas como @willngton observou, habilite `#define TFT_ROTATION`e desejados-o como `TFT_ROTATE_180`já que seu TFT parece estar 180 graus fora.
> 
> Precisaremos de alguém com V1 e V2 para confirmar quais configurações estão corretas e atualizar conforme configurações / leia-me de acordo.

Vou tentar fazer a alteração e aviso se foi resolvido...

Obrigado pela ajuda!

ngoliveira, 2021-09-08:
Fiz a modificação e não funcionou... A tela continua invertida.
Existe outra linha de comando para modificar?
E testei também os movimentos de X,Y e Z, e o movimento do eixo "Z" também está invertido. 
A minha Sapphire possui os cinco Stepers Drivers TCM2208.
[image]
[image]

thinkyhead, 2021-09-09:
> The screen remains inverted.

You can also try all these various options…
```cpp
/**
 * TFT Rotation. Set to one of the following values:
 *
 *   TFT_ROTATE_90,  TFT_ROTATE_90_MIRROR_X,  TFT_ROTATE_90_MIRROR_Y,
 *   TFT_ROTATE_180, TFT_ROTATE_180_MIRROR_X, TFT_ROTATE_180_MIRROR_Y,
 *   TFT_ROTATE_270, TFT_ROTATE_270_MIRROR_X, TFT_ROTATE_270_MIRROR_Y,
 *   TFT_MIRROR_X, TFT_MIRROR_Y, TFT_NO_ROTATION
 */
#define TFT_ROTATION TFT_NO_ROTATION
```
The final result ends up being made with this combination of conditionals…
```cpp
#define TFT_DEFAULT_ORIENTATION (TFT_EXCHANGE_XY | TFT_INVERT_X | TFT_INVERT_Y)
#define TFT_ORIENTATION ((TFT_DEFAULT_ORIENTATION) ^ (TFT_ROTATION))
#define ILI9488_ORIENTATION IF_0((TFT_ORIENTATION) & TFT_EXCHANGE_XY, ILI9488_MADCTL_MV) | \
                            IF_0((TFT_ORIENTATION) & TFT_INVERT_X,    ILI9488_MADCTL_MX) | \
                            IF_0((TFT_ORIENTATION) & TFT_INVERT_Y,    ILI9488_MADCTL_MY)
```

> I also tested the X,Y and Z movements, and the "Z" axis movement is also inverted.

That is an easy fix and well-documented. Take your pick:
```cpp
#define INVERT_Z_DIR false
#define INVERT_Z_DIR true
```

thinkyhead, 2021-10-30:
For best results getting help from smart people, migrate over to Discord… Join link: https://discord.gg/n5NJ59y

closed 2022-04-10 by thinkyhead