MarlinFirmware/Configurations #573: [BUG] (delta generic for BTT skr pro v1.2)

Enzo-Enz, 2021-09-22:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

thisiskeithb, 2021-09-22:
I'm not sure what kind of bug you're reporting since you didn't bother to fill anything out, but you can adapt the generic delta config to any printer: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/delta/generic

closed 2021-09-22 by thisiskeithb

Enzo-Enz, 2021-09-22:
hello guys

I faced with an error while compiling delta generic configuration for BTT skr pro v1.2:

error says:

[image]
[image]
[image]

is there any way to solve that?

thisiskeithb, 2021-09-22:
The error message is telling you to set a `PS_ON_PIN` if you have `PSU_CONTROL` enabled.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Enzo-Enz, 2021-09-22:
thanks for replaying

sorry may I ask you to explain more about it because it is my first time using skr board and and I don't know which part have to be changed for ps on pin

ellensp, 2021-09-22:
The BTT skr pro v1.2 does not have a dedicated pin for PSU_CONTROL.
So if your using that feature you need to add a #define PS_ON_PIN {pin your using as the psu control pin} to your Configuration.h 
Or disable  #define PSU_CONTROL in Configuration.h to turn off the feature by adding // in the front of that line.

Enzo-Enz, 2021-09-22:
thank you so much bro 
the error fixed