MarlinFirmware/Configurations #580: CR10S Pro

akash1769, 2021-10-04:
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `import-2.0.x` branch to see whether the issue still exists. This will require testing with the Marlin `bugfix-2.0.x` branch.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Configurations/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

-->

### Bug Description

Printer won't home on Y. If #define Y_MIN_ENDSTOP_INVERTING set to false, bed moves towards front of machine until a kill is issued. If set to true bed moves towards switch for a moment and then stops. 

### Configuration Files

configuration files come from the example folders here:

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%201.4%20Turbo%20TMC2209

Only baud rate has been changed

### Steps to Reproduce

1. Hit the Y home button on printer display
2. Alternately hit y home button on Ponterface

**Expected behavior:**

I expect the bed to retract back and bounce once on limit switch

**Actual behavior:**

 If #define Y_MIN_ENDSTOP_INVERTING set to false, bed moves towards front of machine until a kill is issued from continuous buzzing indicating trying to find the endstop. If set to true, bed moves towards switch for a moment and then stops. Hitting the y homing button repeatedly moves the bed in the correct direction about 3 cm every time. 

#### Additional Information

The motherboard is a Bigtreetech SKR 1.4 Turbo and TFT35 e.3 V3 

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines
[SKR 1.4 Turbo Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/7275616/SKR.1.4.Turbo.Configuration.zip)
.

ellensp, 2021-10-04:
You should not be using import branch, ever. 
Use the branch that goes with your code
Either bugfix or the release that matches the release code your using.

ellensp, 2021-10-04:
Branches
Release: For most Marlin users. Select the version that matches the Marlin version you are compiling.
Bugfix: For users compiling the bugfix branch of Marlin.
Import: For developers maintaining the Configuration examples.

ellensp, 2021-10-04:
This is a skr 1.4 with tmc2209's, you NEED to cut diag pin off the stepper driver in order to use standard switch based endstops.

closed 2021-10-04 by ellensp

ellensp, 2021-10-04:
[image]

akash1769, 2021-10-05:
ellensp: Thanks for your responses. After using the bugfix version the results were the same. If you read further down in the configuration.h I originally sent you'll see I changed the entries for the drivers from 2209 to 2208. I did the same for the bugfix version. Should I flip the input wire for the stepper motor or is there something else in the config I'm missing?

ellensp, 2021-10-05:
check your endstops are working with M119
If the endstop is activated M119 will report that endstop Triggered, other wise it reports Open.
Test each endstop by moving all axis away from endstop and Sending M119 It should say open
Manually trigger the endstop (hold down switch with finger) and verify M119 now says Triggered.

ellensp, 2021-10-05:
For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).