MarlinFirmware/Configurations #581: [Impossible to define SOL1_PIN] in Marlin version 2.0.9.2

jopl, 2021-10-04:
### Bug Description
There is no possibility to define SOL1_PIN in SOLENOID_PROBE uncommented option.
Despite following definition in "Configuration.h" ( Z Probe Options section):

// A probe that is deployed and stowed with a solenoid pin (SOL1_PIN)
#define SOLENOID_PROBE
#ifdef SOLENOID_PROBE    // jopl: capacitive probe needs power supply (SOL1_PIN) ... 3 lines would be added
#define SOL1_PIN HEATER_1_PIN
#endif

there is an error message:
SanityCheck.h:1508:8: error: #error "SOLENOID_PROBE requires SOL1_PIN."

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.
[Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/7281135/Marlin.zip)

### Steps to Reproduce

1. Using SOL1_PIN as power supply pin for inductive Z probe
2. Uncommenting #define SOLENOID_PROBE
3. Trying to define SOL1_PIN

**Expected behavior:**

No error message during compilation (BTW 2.0.9.1 is OK, but there is problem with homing Y axes with two stepper motors)

SanityCheck.h:1508:8: error: #error "SOLENOID_PROBE requires SOL1_PIN."

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-10-05:
There is a issue. 
The code wont accept there is SOL1_PIN when extruders < 2
With 1 extruder, it will only allow  SOL0_PIN, but then the SOLENOID_PROBE sanity check demands SOL1_PIN...

ellensp, 2021-10-05:
Closing as created on Marlin

closed 2021-10-05 by ellensp