MarlinFirmware/Configurations #590: [BUG] MMU1

AmadoSumbi, 2021-10-16:
### Bug Description

Below Error during compilation.
src\src\MarlinCore.cpp: In function 'void setup()':
src\src\MarlinCore.cpp:1464:15: error: 'mmu_init' was not declared in this scope
     SETUP_RUN(mmu_init());
               ^
src\src\MarlinCore.cpp:1113:53: note: in definition of macro 'SETUP_RUN'
   #define SETUP_RUN(C) do{ SETUP_LOG(STRINGIFY(C)); C; }while(0)
                                                     ^
src\src\MarlinCore.cpp:1464:15: note: suggested alternative: 'HAL_init'
     SETUP_RUN(mmu_init());
               ^
src\src\MarlinCore.cpp:1113:53: note: in definition of macro 'SETUP_RUN'
   #define SETUP_RUN(C) do{ SETUP_LOG(STRINGIFY(C)); C; }while(0)
                                                     ^
Compiling .pio\build\megaatmega2560\src\src\feature\caselight.cpp.o
*** [.pio\build\megaatmega2560\src\src\MarlinCore.cpp.o] Error 1

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.
Below only the modification:
// @section extruder

// This defines the number of extruders
// :[0, 1, 2, 3, 4, 5, 6, 7, 8]
#define EXTRUDERS 4

// Generally expected filament diameter (1.75, 2.85, 3.0, ...). Used for Volumetric, Filament Width Sensor, etc.
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75

// For Cyclops or any "multi-extruder" that shares a single nozzle.
#define SINGLENOZZLE

// Save and restore temperature and fan speed on tool-change.
// Set standby for the unselected tool with M104/106/109 T...
#if ENABLED(SINGLENOZZLE)
  //#define SINGLENOZZLE_STANDBY_TEMP
  //#define SINGLENOZZLE_STANDBY_FAN
#endif

/**
 * Multi-Material Unit
 * Set to one of these predefined models:
 *
 *   PRUSA_MMU1           : Průša MMU1 (The "multiplexer" version)
 *   PRUSA_MMU2           : Průša MMU2
 *   PRUSA_MMU2S          : Průša MMU2S (Requires MK3S extruder with motion sensor, EXTRUDERS = 5)
 *   EXTENDABLE_EMU_MMU2  : MMU with configurable number of filaments (ERCF, SMuFF or similar with Průša MMU2 compatible firmware)
 *   EXTENDABLE_EMU_MMU2S : MMUS with configurable number of filaments (ERCF, SMuFF or similar with Průša MMU2 compatible firmware)
 *
 * Requires NOZZLE_PARK_FEATURE to park print head in case MMU unit fails.
 * See additional options in Configuration_adv.h.
 */
#define MMU_MODEL PRUSA_MMU1

### Steps to Reproduce

1. Uncomment and change MMU2 to MMU1
    #define MMU_MODEL PRUSA_MMU1
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-10-16:
Need to open issue in https://github.com/MarlinFirmware/Marlin/issues as its a marlin bug

ellensp, 2021-10-16:
I have created a PR to fix this, but closing issue as this is not a issue with example configuration files.

closed 2021-10-16 by ellensp