MarlinFirmware/Configurations #593: Hotend temperature reading alarm delay

fghlw, 2021-10-18:
Dear, 
I am runing Marlin , Thermocouple as hotend sensor. But I found that during power on, the first several reading will with errs:  data error or error status. I set mintemp to -9999 can get start sucessful and work. set "THERMOCOUPLE_MAX_ERRORS 50" but seems this only works on serial?  if possible that add a delay for temperayture alarm, that just alarm only never get good result during this "delay time(times)". 
Thank you

fghlw, 2021-10-20:
yes, but this still will close all necessary alarm, set to -9999 at least max alarm can work. and a delay for some non-immediately alarm is normal option.

thinkyhead, 2021-10-30:
Try:
```cpp
#define BOGUS_TEMPERATURE_GRACE_PERIOD 2000 // (ms)
```
If 2000 is too little time, try increasing the value until the error (hopefully) goes away.

fghlw, 2021-10-30:
ok, Thank you

fghlw, 2021-10-31:
works that add "BOGUS_TEMPERATURE_GRACE_PERIOD"

closed 2021-10-31 by fghlw