MarlinFirmware/Configurations #610: Missing

sirdezzy, 2021-11-06:
_Bootscreen and _Statusscreen are missing from the Ender 3 V2 config in both CrealityUI and MarlinUI.

thisiskeithb, 2021-11-06:
They aren't compatible with those UIs.

closed 2021-11-06 by thisiskeithb

The-EG, 2021-11-06:
A bit more detail for those that may come across this with the same thought:
The bootscreen for CrealityUI (and JyersUI & EnhancedUI) is in the DWIN_SET files - `0_start.jpg`. If you want a custom bootscreen, you can change that image and reflash the LCD, as long as it is still the proper format.

The MarlinUI implementation uses icons for the bootscreen since it can be either portrait or landscape mode. If you wanted to customize those, you'd need to customize the appropriate icons and reflash the LCD as above.

All of the UI implementations for this DWIN LCD use various icons in the DWIN_SET assets for the status (and other) screens, so they would be customized as described above.