MarlinFirmware/Configurations #611: [EN] Configuration Request Alfawise U20 one

LRSpfehler, 2021-11-09:
Hallo can anyone Helps me with the Config for the U20 One ?
My problem is that I don't know if I can use the Marlin Config for the U20 for my U20 One. Since I'm a little afraid of destroying my printer with a wrong config, I wanted to ask if anyone here has any experience with the printer.

* Manufacturer: Alfawise
* Model: U20 One
* Motherboard:

thisiskeithb, 2021-11-09:
[There's an Alphawise U20 config available](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Alfawise/U20).

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ellensp, 2021-11-09:
"Alfawise U20 One" seems to be a different from the "Alfawise U20"

tpruvot, 2021-11-09:
indeed, not the same factory for parts, nor the same chip... i guess Alfawise tried to save pennies

ellensp, 2021-11-09:
Notes of what was found in the discord about this machine.

Images of controller https://cdn.discordapp.com/attachments/461605380783472643/907626877009666048/IMG_20211109_144458.jpg
https://cdn.discordapp.com/attachments/461605380783472643/907626877840134264/IMG_20211109_144451.jpg

Motherboard: MPX_ARM_MINI v1.0
Stepper drivers: hr4982mte  
CPU: STM32F103ZET6
TFT Name is: MPX_TFT3.5_V1.0

tpruvot, 2021-11-10:
This motherboard is also in the Alfawise U50 (a recent model too)

tpruvot, 2021-11-10:
ordered an U50 to help you on that, found it at 125€

ellensp, 2021-11-10:
[ALFAWISE-U20-ONE-firmware.zip](https://github.com/MarlinFirmware/Configurations/files/7512084/ALFAWISE-U20-ONE-firmware.zip)
Just binaries, but might have something interesting in them

ellensp, 2021-11-11:
Had a play with the THEME.BIN in the original firmware above, and extracted all the images.
(this is probably not very useful in porting to to current marlin, but was intersting) 

[extracted theme.zip](https://github.com/MarlinFirmware/Configurations/files/7517644/extracted.theme.zip)

There are 132 images 
The largest one being image 65, X= 480 Y= 320
[image]

So I suspect this is the start splash screen and is the full size of the LCD

The script dumpimages.py is in the archive as is a run.log of the output of it being run. 
The log contains information on images sizes, offsets.
(the python is bad... but works)

There is also a image if the machine in the data file# 64, X= 480 Y= 270
[image]

tpruvot, 2021-11-11:
might be already supported by Marlin... see [./Marlin/src/pins/stm32f1/pins_MINGDA_MPX_ARM_MINI.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/stm32f1/pins_MINGDA_MPX_ARM_MINI.h#L34)

top comment to fix in this file, cant be a VET6 with PG pins

tpruvot, 2021-11-11:
[Firmware.U50-2.17.zip](https://github.com/MarlinFirmware/Configurations/files/7522750/Firmware.U50-2.17.zip)

similar...

thinkyhead, 2022-01-26:
Duplicate of #649

closed 2022-01-26 by thinkyhead