MarlinFirmware/Configurations #615: [BUG] (Ender 5 configuration)

axpaul, 2021-11-11:
### Bug Ender 5 configuration files

### Configuration Files

I'm trying to make a new marlin software for the 4.2.7 card on my Ender 5, I removed the previous 1.1 in which I had made a marlin software for the bltouch. I have always used VS Code to generate my files and it works pretty well. Currently, I try to use the default configuration files for the 4.2.7 card provided without making any changes. I get an error message telling me that these files are for an older marlin without details - "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h.".

**Actual behavior:**

Currently I can't build marlin for the 4.2.7 board for the Ender 5

ellensp, 2021-11-11:
You download the wrong version of Config files for the code you are using
See the readme in https://github.com/MarlinFirmware/Configurations (scroll down)

ellensp, 2021-11-11:
look in Marlin/src/inc/Version.h at SHORT_BUILD_VERSION and MARLIN_HEX_VERSION this is the version of Marlin code you are using

ellensp, 2021-11-11:
Also config/README.md  will tell you what code and configs you should use

closed 2021-11-11 by ellensp

axpaul, 2021-11-11:
That's right! 
It works now, thank you !