MarlinFirmware/Configurations #622: [BUG] G28 G428 Error - STOP called because of BLTouch error - restart with M999 / echo:Home XYZ first

aldaw, 2021-11-24:
### Bug Description

When I send the command G28 i get this Error:
Error:!! STOP called because of BLTouch error - restart with M999
After that manual moving the axis don't work.

[image]

And I want to auto home, because M428 for setting new auto home gives me this error:
echo:Home XYZ first

[image]

### Configuration Files
I have commented out:
 #define BLTOUCH_SET_5V_MODE (in Configuration_adv.h)

Configuration.h and Configuration_adv.h
https://www2.zippyshare.com/v/UvLSOFkk/file.html

### Steps to Reproduce
Like descriped above.
**Expected behavior:**
It should auto home and

**Actual behavior:**
That its auto home, and then safe the new set auto-home.

thisiskeithb, 2021-11-24:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-11-24 by thisiskeithb

aldaw, 2021-11-24:
For anyone who encounters the same problems i solved it with commenting out these lines.
#define BLTOUCH_SET_5V_MODE
#define USE_PROBE_FOR_Z_HOMING
#define Z_SAFE_HOMING

The problem is, that i do not use BLTouch and in the configurations archive there is no BLTouch with or without version like on official BTT SKR site.
Commenting those above out did the trick.
Maybe there should be like on the official repo a BLTouch and BLTouch less config. (for Creality CR 10 mini, SKR version)