MarlinFirmware/Configurations #626: [BUG] (PRINTCOUNTER not working)

Hagbard-84, 2021-12-01:
PRINTCOUNTER not working in latest bugfix.

In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\HAL.cpp:30:
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:41:4: warning: #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing." [-Wcpp]
   41 |   #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing."
      |    ^~~~~~~
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:42:4: error: #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
   42 |   #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:41:4: warning: #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing." [-Wcpp]
   41 |   #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing."
      |    ^~~~~~~
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:42:4: error: #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
   42 |   #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:41:4: warning: #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing." [-Wcpp]
   41 |   #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing."
      |    ^~~~~~~
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:42:4: error: #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
   42 |   #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\Sd2Card_sdio_stm32duino.cpp:27:
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:41:4: warning: #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing." [-Wcpp]
   41 |   #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing."
      |    ^~~~~~~
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:42:4: error: #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
   42 |   #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\eeprom_bl24cxx.cpp:32:
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:41:4: warning: #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing." [-Wcpp]
   41 |   #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing."
      |    ^~~~~~~
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:42:4: error: #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
   42 |   #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:41:4: warning: #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing." [-Wcpp]
   41 |   #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing."
      |    ^~~~~~~
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:42:4: error: #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
   42 |   #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
      |    ^~~~~
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\HAL.cpp.o] Error 1
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\HAL_SPI.cpp.o] Error 1
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\eeprom_bl24cxx.cpp.o] Error 1
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\MarlinSerial.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:50,
                 from Marlin\src\HAL\STM32\eeprom_flash.cpp:27:
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:41:4: warning: #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing." [-Wcpp]
   41 |   #warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing."
      |    ^~~~~~~
Marlin\src\HAL\STM32\../../inc/../HAL/STM32/inc/SanityCheck.h:42:4: error: #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
   42 |   #error "Disable PRINTCOUNTER or choose another EEPROM emulation."
      |    ^~~~~
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\Sd2Card_sdio_stm32duino.cpp.o] Error 1
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\Servo.cpp.o] Error 1
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\eeprom_flash.cpp.o] Error 1

ellensp, 2021-12-01:
Not a bug. A limitation in your hardware.
"Disable PRINTCOUNTER or choose another EEPROM emulation."
Chosse one..  either change your type of EEPROM emulation (or add a real eeprom) or disable  PRINTCOUNTER

closed 2021-12-01 by ellensp

Hagbard-84, 2021-12-01:
is it posible too edit the size of eeprom?  its a STM32F401MCU

Hagbard-84, 2021-12-01:
the botloader: do i need to define in merlin ware i putt the firmware? if i build it from latest bugfix.

ellensp, 2021-12-01:
This is not a general questions support forum.
This is the Marlin githug for pre-configured example configurations

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y