MarlinFirmware/Configurations #627: [BUG] As of 10 hours ago, all Configuration_adv.h files are broken

ellensp, 2021-12-05:
The error is:  Configuration_adv.h:1336:2: error: #endif without #if

This is the cause

```CPP
  #endif

    #if ENABLED(AUTO_BED_LEVELING_BILINEAR)
      // Add a calibration procedure in the Probe Offsets menu
      // to compensate for twist in the X-axis.
      //#define X_AXIS_TWIST_COMPENSATION
      #if ENABLED(X_AXIS_TWIST_COMPENSATION)
        /**
         * Enable to init the Probe Z-Offset when starting the Wizard.
         * Use a height slightly above the estimated nozzle-to-probe Z offset.
         * For example, with an offset of -5, consider a starting height of -4.
         */
        #define XATC_START_Z 0.0
        #define XATC_MAX_POINTS 3             // Number of points to probe in the wizard
        #define XATC_Y_POSITION Y_CENTER      // (mm) Y position to probe
      #endif
    #endif
  #endif
```

The first #endif needs deleted.

thisiskeithb, 2021-12-05:
Fixed in a912d4f.

thisiskeithb, 2021-12-05:
Duplicate of #627

closed 2021-12-05 by thisiskeithb