MarlinFirmware/Configurations #638: Incorrect driver type for Ender3 V2 4.2.2 in Marlin 2.0.9.3

dmitrygribenchuk, 2021-12-26:
Hello. I have an Ender 3v2 (4.2.2 version) and try to update from 2.0.9.2 to 2.0.9.3.
When I build new firmware I have a notice that Marlin/src/inc/Warnings.cpp:537:4: warning: #warning "Creality 4.2.2 boards may have A4988 or TMC2208_STANDALONE drivers. Check your board and make sure to select the correct DRIVER_TYPE!" [-Wcpp]

And after the firmware update, I have a black screen on my printer.

It's reproduced on CrealityV422 CrealityUI and MarlinUI config files.

ellensp, 2021-12-28:
The 4.2.2 comes with 5 different stepper driver types
There is no way to detect this. You have you physically look at your controller for the hand written letter written on the sdcard cover. It will be one of these letters
C = HR4988
E = A4988
A = TMC2208
B = TMC2209
H = TMC2225
Thus you get an warning "Check your board and make sure to select the correct DRIVER_TYPE!"
  
The board not booting is unrelated to this.
There was an issue with  2.0.9.3. that stopped most boards starting up. This has nee resolved
Get a fresh copy of  2.0.9.3. and try again

closed 2021-12-28 by ellensp

dmitrygribenchuk, 2021-12-28:
Thanks a lot. I'm rebuilding new firmware from bugfix-2.0.x branch and its works.

3dfactor, 2022-01-09:
> The 4.2.2 comes with 5 different stepper driver types There is no way to detect this. You have you physically look at your controller for the hand written letter written on the sdcard cover. It will be one of these letters C = HR4988 E = A4988 A = TMC2208 B = TMC2209 H = TMC2225 Thus you get an warning "Check your board and make sure to select the correct DRIVER_TYPE!"
> 
> The board not booting is unrelated to this. There was an issue with 2.0.9.3 and bugfix that stopped most boards starting up. This has been resolved Get a fresh copy of 2.0.9.3. and try again

I'm sorry to revive a two week old thread, however I didn't want to pollute the issues list with new issue of basically the same thing. My issue is that my boards sdcard slot has "RC" written on it. Do I follow the C and go with HR4988 driver option or does the "R" have any significance in deciphering it?

[image]

r0dnay, 2022-01-16:
i believe the R will be irrelevant because there is no R type in the list
im wondering where and in which file this needs to be specified?

r0dnay, 2022-01-17:
> The 4.2.2 comes with 5 different stepper driver types There is no way to detect this. You have you physically look at your controller for the hand written letter written on the sdcard cover. It will [...]

any idea where DRIVER_TYPE is defined?

3dfactor, 2022-01-17:
> i believe the R will be irrelevant because there is no R type in the list im wondering where and in which file this needs to be specified?

Actually I already found out what **RC** is - it's the label of Creality scamming their clients. RC stands for STM32F103**RCT6** microprocessor which is inferior to the originally sold boards with STM32F103**RET6** . This microprocessor has half the FLASH and SRAM memory...
I've changed my board to a proper 4.2.7 since I found that out.

vkhodygo, 2022-03-14:
@3dfactor You're a saviour, I've been looking for a solution for days. I had to take the whole board apart anyway, but that's a different story. 

Those poor folks that don't know what to do, read the text below:
There is a configuration for such boards (mine is all TMC2208s, there are other similar reports), but as of today it exists only in the `bugfix-2.0.x` branch (both config and firmware). I also refer to a brief Reddit [discussion](https://www.reddit.com/r/ender3v2/comments/slqgnb/creality_ender_3_v2_board_422_arm_stm32f103rct6/) regarding this issue, where @brunohenriquy suggests their own solution based on a fork.

AkashicSeer, 2023-01-19:
My Ender 3 max was supposed to come with all silent drivers. I have an A/C on my board. I am guessing, since reading the configuration.h file that this means I got ripped off and got 2 2208 drivers and 2 4988.  Why is honesty so hard for Chinese companies?. Do they not understand that when you screw people they will move to one of your competitors instead? Is that a hard concept to grasp?  This mixing of chips and keeping the same damn board version is not intelligent in any way. You don't change shit like chipsets and leave the damned version the same. Doing so causes all of the damned confusion we see with creality 4.2.2 which is a POS

 I call it the forrest gump frankenboard. Because it is a mix of bullshit and you never know what you got. In order to know I must risk destroying the board by removing a glued on heatsink???? This makes updating the firmware a real pain.  Creality sucks. What a bunch of dishonest liars. I hope they don't think I'll buy anymore of their fail shit. When I replace this MB I'll go with one of their competitors and replace the screen and MB at the same time.  Nothing pisses myself and other westerners off more than dishonesty and getting ripped off.  Why would I give a liar more of my money, after they scammed me once? For all I know I'll get another damned frankenboard FFS. I hope they enjoy the money they got from lying to people, because it won't pay off in the end. They will learn the hard way that Dishonesty is not the best policy. 
 
 I hate liars and dishonesty more than anything else. It makes my blood boil when people think I am stupid and gullible and that is what people are saying when they lie to you "Oh this person is an idiot, they'll never know I f-ed them,"  Oh I will know and I will not STFU about it.