MarlinFirmware/Configurations #640: Anycubic Chiron Config 2.0.9.3 build fails. Undefined pin TG_FAN1_PIN

SwiftNick, 2021-12-27:
The recent refactoring of controller pin definitions has broken the config file by removing the definition of TG_FAN1_PIN.

[Config_adv.h Line 511](https://github.com/MarlinFirmware/Configurations/blob/aa05f74b93c2615d3ece9d53496bc00268509426/config/examples/AnyCubic/Chiron/Configuration_adv.h#L511)

A simple fix just change TG_FAN1_PIN to FAN1_PIN

stuartBainbridge, 2021-12-27:
> The recent refactoring of controller pin definitions has broken the config file by removing the definition of TG_FAN1_PIN.
>
> [Config_adv.h Line [...]

Noticed this too.

ellensp, 2021-12-27:
It used to be a custom pin name, ie #define TG_FAN1_PIN 7
Looks like it has been updated to use standard names so Configuration_adv.h should be #define CONTROLLER_FAN_PIN FAN1_PIN

ellensp, 2021-12-27:
Please test and confirm, then we can create a PR to fix. this

thisiskeithb, 2021-12-27:
> Looks like it has been updated to use standard names so Configuration_adv.h should be #define CONTROLLER_FAN_PIN FAN1_PIN

See https://github.com/MarlinFirmware/Marlin/pull/22983

SwiftNick, 2021-12-27:
> Please test and confirm, then we can create a PR to fix. this

I have tested this on a Chiron and the fix is fine.

stuartBainbridge, 2021-12-27:
This has been fine for me also.

ellensp, 2021-12-28:
Pr to fix this has been created,

ellensp, 2022-02-11:
@thinkyhead
https://github.com/MarlinFirmware/Configurations/pull/642 needs to be merged into release 2.0.9.3 also  
See https://github.com/MarlinFirmware/Marlin/issues/23695

thisiskeithb, 2022-04-09:
I'm gonna go out on a limb and say that 2.0.9.3 won't be retagged for this issue, but it's been fixed in the latest bugfix and will be fixed once 2.1 is tagged & released.

closed 2022-04-09 by thisiskeithb